gnumpi/esphome_audio #12: Onju Home Mic and Speaker not working

sqldiablo, 2024-02-25:
Hi there,

I used the example yaml file in /examples to modify my working yaml file for my Onju home speaker. Since Onju home is based on the same esp32-s3-devkitc-1 as the example yaml uses, I was hoping that this would work for me, however, the device can't seem to hear anything, and I get an error when I try to play audio through the media player. My yaml and logs are below. I've tried the left and right channels for the mic, but neither work. I also tried it with or without the noise suppression, auto name, and volume multiplier settings, but I get the same results with or without those settings. I'm wondering if this is related to needing external ADC/DAC support, but I'm not sure. Would you be able to help me with this? I'm happy to help debug however I can.

YAML file:
```yaml
substitutions:
  name: "living-room-onju-home"
  friendly_name: "Living Room Onju Home"
  wifi_ap_password: "<REDACTED>"

external_components:
  - source:
      type: git
      url: https://github.com/gnumpi/esphome_audio
      ref: main
    components: [ adf_pipeline, i2s_audio ]

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  min_version: 2024.2.0
  platformio_options:
    build_flags: "-DBOARD_HAS_PSRAM"
    board_build.arduino.memory_type: qio_opi
  on_boot:
    then:
      - light.turn_on:
          id: top_led
          effect: slow_pulse
          red: 100%
          green: 60%
          blue: 0%
      - wait_until:
          condition:
            wifi.connected:
      - light.turn_on:
          id: top_led
          effect: pulse
          red: 0%
          green: 100%
          blue: 0%
      - wait_until:
          condition:
            api.connected:
      - light.turn_on:
          id: top_led
          effect: none
          red: 0%
          green: 100%
          blue: 0%
      - delay: 1s
      - script.execute: reset_led

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf
    version: recommended
    sdkconfig_options:
      # need to set a s3 compatible board for the adf-sdk to compile
      # board specific code is not used though
      CONFIG_ESP32_S3_BOX_BOARD: "y"

psram:
  mode: octal
  speed: 80MHz

logger:
[... 663 lines cut ...]

[19:37:58][D][esp-idf:000]: E (69980) AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore.
Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_4.4._freertos.patch" first

[19:37:58][D][esp-idf:000]: E (69982) AUDIO_THREAD: Error creating RestrictedPinnedToCore decoder

[19:37:58][D][esp-idf:000]: E (69984) AUDIO_ELEMENT: [decoder] audio_thread_create failed

[19:37:58][E][esp_audio_sources:048]: Starting decoder streamer failed
[19:37:58][D][esp-idf:000]: W (69988) AUDIO_ELEMENT: [http] Element has not create when AUDIO_ELEMENT_RESUME

[19:37:58][E][esp_audio_sources:053]: Resuming http streamer failed
[19:37:58][D][esp-idf:000]: W (69991) AUDIO_ELEMENT: [decoder] Element has not create when AUDIO_ELEMENT_RESUME

[19:37:58][E][esp_audio_sources:056]: Resuming decoder failed
[19:37:58][I][esp_audio_sources:058]: Streamer status: 7
[19:37:58][I][esp_audio_sources:059]: decoder status: 7
[19:37:58][I][esp_adf_pipeline:025]: Starting request, current state 5
[19:37:58][I][adf_audio:134]: got new pipeline state: 1
[19:37:58][I][esp_adf_pipeline:098]: set_new_state: 1
[19:37:58][I][esp_adf_pipeline:245]: [ * ] CMD: 8 Pipeline: 1
[19:37:58][I][esp_adf_pipeline:250]: [ http ] status: 1
[19:37:58][I][esp_adf_pipeline:245]: [ * ] CMD: 8 Pipeline: 1
[19:37:58][I][esp_adf_pipeline:250]: [ decoder ] status: 1
```

sqldiablo, 2024-02-25:
More details on Onju Home, in case that is helpful: https://github.com/justLV/onju-voice

gnumpi, 2024-02-25:
Your log says that the idf_4.4._freertos.patch is needed but not correctly installed.
```
...
Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_4.4._freertos.patch" first
...
```
Probably I didn't run into this problem because I was using a project based on https://github.com/esphome/esphome/pull/5230 first, which applies the necessary patch and I guess the patch has to be applied only once? Can anyone confirm this? 

I opened the issue #13 to address the patching.

gnumpi, 2024-02-25:
I copied the code for applying the patch from https://github.com/esphome/esphome/pull/5230. 
Could you try running the adf_pipeline component from the 13-apply-idf_v44_freertospatch-at-build-time branch and see if it is of any help?

```yaml
- source:
     type: git
     url: https://github.com/nighi/esphome_audio
     ref: 13-apply-idf_v44_freertospatch-at-build-time
   components: [ adf_pipeline, i2s_audio ]
   refresh: 0s
```

sqldiablo, 2024-02-25:
This has definitely been a step in the right direction. The media player is now working and I can play audio through it. However, the microphone doesn't seem to be starting. On the original yaml for this, it was using an external ADC, so I wonder if something is wrong with ADF connecting to the external ADC.

yaml config:
```yaml
substitutions:
  name: "living-room-onju-home"
  friendly_name: "Living Room Onju Home"

external_components:
  - source:
      type: git
      url: https://github.com/gnumpi/esphome_audio
      ref: 13-apply-idf_v44_freertospatch-at-build-time
    components: [ adf_pipeline, i2s_audio ]

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  min_version: 2024.2.0
  platformio_options:
    build_flags: "-DBOARD_HAS_PSRAM"
    board_build.arduino.memory_type: qio_opi
  on_boot:
    then:
      - light.turn_on:
          id: top_led
          effect: slow_pulse
          red: 100%
          green: 60%
          blue: 0%
      - wait_until:
          condition:
            wifi.connected:
      - light.turn_on:
          id: top_led
          effect: pulse
          red: 0%
          green: 100%
          blue: 0%
      - wait_until:
          condition:
            api.connected:
      - light.turn_on:
          id: top_led
          effect: none
          red: 0%
          green: 100%
          blue: 0%
      - delay: 1s
      - script.execute: reset_led

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf
    version: recommended
    sdkconfig_options:
      # need to set a s3 compatible board for the adf-sdk to compile
      # board specific code is not used though
      CONFIG_ESP32_S3_BOX_BOARD: "y"

psram:
  mode: octal
  speed: 80MHz

logger:
api:
  encryption:
    key: <REDACTED>
[... 2715 lines cut ...]
[10:53:36][D][light:036]: 'top_led' Setting:
[10:53:36][D][light:051]:   Brightness: 100%
[10:53:36][D][light:059]:   Red: 100%, Green: 0%, Blue: 100%
[10:53:36][D][light:085]:   Transition length: 1.0s
[10:53:49][D][media_player:059]: 'Living Room Onju Home' - Setting
[10:53:49][D][media_player:066]:   Media URL: http://<REDACTED>:8123/api/tts_proxy/a54d88e06612d820bc3be72877c74f257b561b19_en-gb_8cd8d30e6e_tts.piper.mp3
[10:53:50][W][component:214]: Component i2s_audio.media_player took a long time for an operation (0.52 s).
[10:53:50][W][component:215]: Components should block for at most 20-30ms.
[10:53:50][W][component:214]: Component i2s_audio.media_player took a long time for an operation (0.06 s).
[10:53:50][W][component:215]: Components should block for at most 20-30ms.
[10:53:50][W][component:214]: Component i2s_audio.media_player took a long time for an operation (0.06 s).
[10:53:50][W][component:215]: Components should block for at most 20-30ms.
[10:53:50][W][component:214]: Component i2s_audio.media_player took a long time for an operation (0.06 s).
[10:53:50][W][component:215]: Components should block for at most 20-30ms.
[10:53:51][W][component:214]: Component i2s_audio.media_player took a long time for an operation (0.06 s).
[10:53:51][W][component:215]: Components should block for at most 20-30ms.
[10:53:51][W][component:214]: Component i2s_audio.media_player took a long time for an operation (0.06 s).
[10:53:51][W][component:215]: Components should block for at most 20-30ms.
[10:53:51][W][component:214]: Component i2s_audio.media_player took a long time for an operation (0.06 s).
[10:53:51][W][component:215]: Components should block for at most 20-30ms.
[10:53:52][W][component:214]: Component i2s_audio.media_player took a long time for an operation (0.51 s).
[10:53:52][W][component:215]: Components should block for at most 20-30ms.
[10:53:52][W][component:214]: Component i2s_audio.media_player took a long time for an operation (0.51 s).
[10:53:52][W][component:215]: Components should block for at most 20-30ms.
```

gnumpi, 2024-02-27:
Unfortunately, using the same i2s controller for the microphone and also for the speaker is not working yet. I opened an issue #17 for this and hope that I can add the duplex feature soon.

sqldiablo, 2024-02-27:
Thanks for the update. I appreciate your help so far, and I'll keep an eye on this repo for if/when you're able to get to it.

gnumpi, 2024-04-18:
This issue has been solved in https://github.com/gnumpi/esphome_audio/issues/17, I will close this issue.

closed 2024-04-18 by gnumpi