gnumpi/esphome_audio #27: Offering help if needed

X-Ryl669, 2024-04-12:
Hi,

 I've been linked to your repository via [here](https://github.com/esphome/firmware/pull/173#issuecomment-2051222537). 
I've implemented an update to ESP-ADF component to support the latest version and sharing I2C bus with ESP-ADF's format and ESPHome. My work is [here](https://github.com/X-Ryl669/esphome/tree/esp-adf-take2

I think it can be useful to merge my changes (at least for using the last ESP-ADF and dependencies version). There were a lot of pitfalls I had to fix (like the esp-dsp dependency, the I2C issue, the esp-adf incompatibility with C and C++ structure definitions and so on). So it might be save you a lot of time to avoid redoing this.

Let me know if you need it.

gnumpi, 2024-04-17:
Hey, sounds really interesting. Do I get it right that supporting the ADF-I2C let you use the DACs and ADCs supported by the ADF-boards? I also added some DAC and ADC support to the I2S_audio component in this repository. Would be awesome to add the ADF-boards right away. Could you share a config which makes use of your change to get a better understanding?

gnumpi, 2024-04-18:
Really great news! I will try to integrate the [esp_codec_dev](https://github.com/espressif/esp-adf/tree/master/components/esp_codec_dev) devices into the `adf_pipeline` component then. 
Thanks for offering your help! If it's ok, I will come back to you when I have a rough beta version and see what you think.