gnumpi/esphome_audio #29: Unable to Compile when using adf_pipeline

stevetheman1031, 2024-04-19:
I've run into a bit of a snag with compiling ESPHome when I add the adf_pipeline. Strangely, I can compile just fine when I comment out the adf_pipeline section. I'm a bit puzzled about why I'm encountering this error, especially since I'm not even using the touch_sensor component.

You can find my configuration here: https://pastebin.com/ZPjWaY06

And here's the compilation error I'm encountering: https://pastebin.com/VWbYA4tn

I'd appreciate any insights or assistance you might have on resolving this issue. Thanks in advance for your help!

cowboyrushforth, 2024-04-19:
try to add this to your config, under esp32.framework

```
    sdkconfig_options:
      # need to set a s3 compatible board for the adf-sdk to compile
      # board specific code is not used though
      CONFIG_ESP32_S3_BOX_BOARD: "y"
```

otherwise it seems to try to include other boards

gnumpi, 2024-04-19:
Thanks for pointing at it, I forgot to put this info into the README after rewriting it.

closed 2024-07-05 by gnumpi