gnumpi/esphome_audio #31: Instability issues

dinki, 2024-04-19:
Hey guys.  Thanks for posting this.  I'm excited to be able to use the esp32 as a voice assistant with media player.  I had a lot of help getting this going and don't know a whole lot about this so excuse my ignorance.  A little backstory:

This is the original config I used without issue as provided by BigBobba on discord:
https://dpaste.org/1E3fb

I asked about media_player and he helped getting me going with this:
https://dpaste.org/wEyKH

Since switching I have experienced nothing but instability.  I thought it might be hardware related so I wired up a separate esp32 with different DAC and mic and flashed the same bin file as the previous and got the same result.  I changed the config back to the first above and it works without issue.

The media player did play fine initially for probably several hours but then got in this weird state where it appeared to be rebooting randomly and when I tried calling it with the wake word.  Here's some logs if it helps:

https://dpaste.org/ESfwp

Any help is most appreciated.  Thanks.

dinki, 2024-04-20:
Just reflashed and using a 3A power supply instead of the two separate 1A power supplies I was using previously.  Everything else is the same.  So far all is well.  Will let it ride for a bit and see if that was the problem.  Will close out in the next few days if the problem stays away.

scottcopus, 2024-04-20:
When using separate power supplies just make sure each one’s “grounds” are
tied together.

Cheers,
Scott

dinki, 2024-04-20:
Thanks for the tip.  For now these are just going back to the ESP32 with dupont connectors.  All components using ground on the board.

dinki, 2024-04-20:
https://dpaste.org/dtbu4

Spoke too soon.  It is back in a weird state.  I do not know why.  I can say that it does detect wake word some times but others not.  It does not respond at all now.  I have tried restarting the chip as well as reloading in HA.  I am also seeing the RGB led flash periodically too.

I am running an Aliexpress S3 device.  Who knows if it is of questionable quality.  Is there a recommended model with purchase link?  I am, as always, on a budget but do not mind spending a bit more for something confirmed to be good if you guys think this might be the case.

gnumpi, 2024-04-20:
Sorry to hear that you encounter issues. Actually I found a problem when the pipeline is stopped during the preparation phase, which I am trying to fix right now. Anyhow, I wonder why your media player is called multiple times! You are calling the media player again while the MP is still in preparation phase thats why you run into the same issue I just found. Do you have an idea why your media player is called that often?

dinki, 2024-04-21:
Please excuse my ignorance but I'm not sure.  I can tell you that I was not trying to do this intentionally.  I think that this same behavior will happen again as soon as I plug the device in again.  I can check the logs again tomorrow.  I really find it odd that this media is trying to play every time I power the device on.  

I know previously I saw an alarm mp3 file I was using for testing my assist alarm routine.  This was repeated multiple times just like this one.  I'm pretty certain I restarted HA and the same mp3 was still repeatedly played.  

Do know that I am playing some of this media using Music Assistant.  If unfamiliar, MA connects to HA and creates its own media_player.  I really don't know how this works but the MA device somehow plays its stream through that device to the HA media_player device.  No clue if that is causing issues.

Please let me know if what you are chasing down may be what I am experiencing as well.

pimp1310, 2024-05-06:
@gnumpi 

is my problem with the media_player related to the same problem in this thread?

https://github.com/gnumpi/esphome_audio/issues/36

gnumpi, 2024-05-07:
You can try the fixes that helped resolve my similar issues:
```yaml
- source:
     type: git
     url: https://github.com/gnumpi/esphome_audio
     ref: dev-next
   components: [ adf_pipeline, i2s_audio ]
```