gnumpi/esphome_audio #32: Media Player not working

andreasbrett, 2024-04-26:
First of all a huge thank you providing the adf component. I'm currently trying to create a HA voice satellite with microwakeword with an ESP32 S3 devkit (NR16R8), an INMP441 microphone and a 3W speaker connected to the MAX98357 DAC. So basically the same setup as in your examples.

Inputs work really well but I just can't get the speaker output to work. I'm getting the following error when trying to run the demo sound from espressif (in your examples the "Pipeline" switch).

```
[13:16:57][D][media_player:059]: 'Speaker' - Setting
[13:16:57][D][media_player:066]:   Media URL: https://dl.espressif.com/dl/audio/ff-16b-2c-44100hz.mp3
[13:16:57][D][adf_media_player:030]: Got control call in state 1
[13:16:57][D][esp_adf_pipeline:050]: Starting request, current state UNINITIALIZED
[13:16:57][V][esp-idf:000]: I (106611) MP3_DECODER: MP3 init
[13:16:57]
[13:16:57][V][esp-idf:000]: I (106614) I2S: DMA Malloc info, datalen=blocksize=2048, dma_buf_count=4
[13:16:57]
[13:16:57][D][i2s_audio:072]: Installing driver : yes
[13:16:57][D][esp_adf_pipeline:358]: pipeline tag 0, http
[13:16:57][D][esp_adf_pipeline:358]: pipeline tag 1, decoder
[13:16:57][D][esp_adf_pipeline:358]: pipeline tag 2, i2s_out
[13:16:57][V][esp-idf:000]: I (106619) AUDIO_PIPELINE: link el->rb, el:0x3d820e94, tag:http, rb:0x3d821390
[13:16:57]
[13:16:57][V][esp-idf:000]: I (106620) AUDIO_PIPELINE: link el->rb, el:0x3d82104c, tag:decoder, rb:0x3d8223d0
[13:16:57]
[13:16:57][D][esp_adf_pipeline:368]: Setting up event listener.
[13:16:57][D][esp_adf_pipeline:302]: State changed from UNINITIALIZED to PREPARING
[13:16:57][I][adf_media_player:135]: got new pipeline state: 1
[13:16:57][D][adf_i2s_out:127]: Set final i2s settings: 16000
[13:16:57][V][esp-idf:000]: I (106642) AUDIO_THREAD: The http task allocate stack on external memory
[13:16:57]
[13:16:57]
[13:16:57]
[13:16:57][V][esp-idf:000]: E (106647) AUDIO_THREAD: Error creating RestrictedPinnedToCore http
[13:16:57]
[13:16:57][V][esp-idf:000]: E (106648) AUDIO_ELEMENT: [http] audio_thread_create failed
[13:16:57]
[13:16:57][E][esp_audio_sources:085]: Starting http streamer failed
[13:16:57][V][esp-idf:000]: I (106650) AUDIO_THREAD: The decoder task allocate stack on external memory
[13:16:57]
[13:16:57]
[13:16:57]
[13:16:57][V][esp-idf:000]: E (106654) AUDIO_THREAD: Error creating RestrictedPinnedToCore decoder
[13:16:57]
[13:16:57][V][esp-idf:000]: E (106656) AUDIO_ELEMENT: [decoder] audio_thread_create failed
[13:16:57]
[13:16:57][E][esp_audio_sources:088]: Starting decoder streamer failed
[13:16:57][V][esp-idf:000]: W (106658) AUDIO_ELEMENT: [http] Element has not create when AUDIO_ELEMENT_RESUME
[13:16:57]
[13:16:57][E][esp_audio_sources:092]: Resuming http streamer failed
[13:16:57][V][esp-idf:000]: W (106660) AUDIO_ELEMENT: [decoder] Element has not create when AUDIO_ELEMENT_RESUME
[13:16:57]
[13:16:57][E][esp_audio_sources:095]: Resuming decoder failed
```

TBH these error messages are rather cryptic to me. I have no clue why there are errors in fetching the mp3 via http(s). Do you have any idea, where I could be wrong?

Here's my config: [voice-satellite.yaml.txt](https://github.com/gnumpi/esphome_audio/files/15129961/voice-satellite.yaml.txt)

gnumpi, 2024-04-26:
Which ESPHome version do you use and which IDF version is getting installed? Maybe the idf_v4.4_freertos.patch did not get installed correctly. It should get installed the first time you include the adf_pipeline component.

imonlinux, 2024-04-30:
@andreasbrett 
I don't know which NR16R8 board you are using, but I was troubleshooting the MAX98357A only working on the 3.3v pin when I ran across this thread:
https://forum.arduino.cc/t/chinese-esp32-s3-5v-pin-warning/1192758

Once I bridged the IN-OUT pads, the 5v pin started making 5v and my speaker started working.

scottcopus, 2024-04-30:
Yikes. Guess it’s a good idea fall back to getting out your multimeter at
the first sign of something not working and checking voltages especially at
all power pins of external modules and boards.

andreasbrett, 2024-04-30:
> @andreasbrett I don't know which NR16R8 board you are using, but I was troubleshooting the MAX98357A only working on the 3.3v pin when I ran across this thread: [...]

Thank you but I had to figure that out already (same for the onboard RGB LED that required bridging - love when products come without documentation and weird default setups). Unfortunately this was not the culprit. Meanwhile I transferred my ESPHome instance to a different machine because building the firmware took so long. With a fresh build I now can get the media player to play mp3 files via http and https (particularly the "wake word detected" beep). It's still not working though with the TTS proxied response mp3 files from Nabu Casa. I will have to figure out what's different with those https requests. Maybe something weird with my TLS setup at home.

andreasbrett, 2024-04-30:
> Yikes. Guess it’s a good idea fall back to getting out your multimeter at the first sign of something not working and checking voltages especially at all power pins of external modules and boards.
> […](#)

You have no clue how spot on you are! I had ordered 2 cheap MAX98357A from Amazon. When I figured out that on my S3 board the 5V pin was Vin by default and connected the bridge to make it a Vout, suddenly the DAC got really hot and it smelled a bit smokey :) Consider me shocked when I found out, that on BOTH DACs the Vin and Gnd pins were shorted! I looked up comments on Amazon and I'm not the only one. I guess where back in the age of "don't trust nobody, measure yourself".

andreasbrett, 2024-04-30:
I updated to ESPHome 2024.4.2 and Home Assistant 2024.5.b4 and rebuilt the firmware. Now everything works as expected. I get TTS responses from Assist and can also play TTS through the media player component. Not sure which changes exactly contributed to the errors not occuring again.

closed 2024-04-30 by andreasbrett