gnumpi/esphome_audio #33: unable to compile -  Unable to find current revision in submodule path 'components/esp-adf-libs

pimp1310, 2024-04-28:
hello,

i tried to compile but i always get  a error, i dont know what exactly he wants. i think he cant fetch the esp-adf component?

`INFO ESPHome 2024.4.1
INFO Reading configuration /config/esphome/v-assist-kitchen-esp32nr168.yaml...
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf.git@v2.5
INFO Updating submodules (components/esp-adf-libs, components/esp-sr) for https://github.com/espressif/esp-adf.git@v2.5
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1061, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1048, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 478, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 193, in write_cpp
    return write_cpp_file()
           ^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 211, in write_cpp_file
    writer.write_cpp(code_s)
  File "/esphome/esphome/writer.py", line 344, in write_cpp
    copy_src_tree()
  File "/esphome/esphome/writer.py", line 297, in copy_src_tree
    copy_files()
  File "/esphome/esphome/components/esp32/__init__.py", line 685, in copy_files
    repo_dir, _ = git.clone_or_update(
                  ^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/git.py", line 111, in clone_or_update
    run_git_command(
  File "/esphome/esphome/git.py", line 31, in run_git_command
    raise cv.Invalid(lines[-1][len("fatal: ") :])
voluptuous.error.Invalid: Unable to find current revision in submodule path 'components/esp-adf-libs'`

here is my config

https://pastebin.com/4KuYS7GG

Thank You!

gnumpi, 2024-04-29:
Hey, sorry I can't reproduce your error. It seems something is wrong with your general build setup. Can you retry with a fresh setup?

pimp1310, 2024-04-29:
@gnumpi 

You mean reinstall esphome ? 
Or only the sketch?

pimp1310, 2024-05-04:
@gnumpi 

now it works. i reinstalled esphome addon.

but now he doesnt listen to my commands, he hears the wake word but not more

`[15:28:01][D][esp_adf_pipeline:302]: State changed from RUNNING to STOPPING
[15:28:01][D][esp-idf:000]: W (50270) AUDIO_ELEMENT: OUT-[i2s_in] AEL_IO_ABORT

[15:28:01][D][esp-idf:000]: W (50272) AUDIO_ELEMENT: OUT-[i2s_in] AEL_IO_ABORT

[15:28:01][D][esp-idf:000]: W (50274) AUDIO_ELEMENT: OUT-[i2s_in] AEL_IO_ABORT

[15:28:01][D][esp_adf_pipeline:400]: Called deinit_all
[15:28:01][D][esp-idf:000]: I (50285) AUDIO_PIPELINE: audio_pipeline_unlinked

[15:28:01][D][esp-idf:000]: W (50287) AUDIO_ELEMENT: [i2s_in] Element has not create when AUDIO_ELEMENT_TERMINATE

[15:28:01][D][esp-idf:000]: W (50302) AUDIO_ELEMENT: [pcm_reader] Element has not create when AUDIO_ELEMENT_TERMINATE

[15:28:01][D][esp-idf:000]: I (50305) I2S: DMA queue destroyed

[15:28:01][D][esp_adf_pipeline:302]: State changed from STOPPING to UNINITIALIZED
[15:28:01][D][media_player:059]: 's3-dev_media_player' - Setting
[15:28:01][D][media_player:063]:   Command: STOP
[15:28:01][D][esp_adf_pipeline:085]: Called 'stop' while in UNINITIALIZED state.
[15:28:01][W][light:476]: 'Status led' - No such effect 'Slow Pulse'
[15:28:01][D][light:036]: 'Status led' Setting:
[15:28:01][D][light:047]:   State: ON
[15:28:01][D][voice_assistant:439]: State changed from IDLE to START_PIPELINE
[15:28:01][D][voice_assistant:445]: Desired state set to START_MICROPHONE
[15:28:01][W][component:237]: Component micro_wake_word took a long time for an operation (540 ms).
[15:28:01][W][component:238]: Components should block for at most 30 ms.
[15:28:01][D][voice_assistant:126]: microphone not running
[15:28:01][D][voice_assistant:210]: Requesting start...
[15:28:01][D][voice_assistant:439]: State changed from START_PIPELINE to STARTING_PIPELINE
[15:28:01][D][voice_assistant:126]: microphone not running
[15:28:01][D][voice_assistant:126]: microphone not running
[15:28:01][D][voice_assistant:126]: microphone not running
[15:28:01][D][voice_assistant:476]: Client started, streaming microphone
[15:28:01][D][voice_assistant:439]: State changed from STARTING_PIPELINE to START_MICROPHONE
[15:28:01][D][voice_assistant:445]: Desired state set to STREAMING_MICROPHONE
[15:28:01][D][voice_assistant:163]: Starting Microphone
[15:28:01][D][esp_adf_pipeline:050]: Starting request, current state UNINITIALIZED
[15:28:01][D][esp-idf:000]: I (50937) I2S: DMA Malloc info, datalen=blocksize=1024, dma_buf_count=4

[15:28:01][D][i2s_audio:072]: Installing driver : yes
[15:28:01][D][esp_adf_pipeline:358]: pipeline tag 0, i2s_in
[15:28:01][D][esp_adf_pipeline:358]: pipeline tag 1, pcm_reader
[15:28:01][D][esp-idf:000]: I (50965) AUDIO_PIPELINE: link el->rb, el:0x3d8263e0, tag:i2s_in, rb:0x3d8266bc

[15:28:01][D][esp_adf_pipeline:370]: Setting up event listener.
[15:28:01][D][esp_adf_pipeline:302]: State changed from UNINITIALIZED to PREPARING
[15:28:01][D][voice_assistant:439]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[15:28:01][W][component:237]: Component voice_assistant took a long time for an operation (68 ms).
[15:28:01][W][component:238]: Components should block for at most 30 ms.
[15:28:01][D][esp_adf_pipeline:302]: State changed from PREPARING to STARTING
[15:28:01][D][esp-idf:000]: I (51017) AUDIO_ELEMENT: [i2s_in-0x3d8263e0] Element task created

[15:28:02][D][esp-idf:000]: I (51027) AUDIO_ELEMENT: [pcm_reader-0x3d826574] Element task created

[15:28:02][D][esp-idf:000]: I (51037) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:8364355 Bytes, Inter:135816 Bytes, Dram:135816 Bytes

[15:28:02][D][esp-idf:000]: I (51047) AUDIO_ELEMENT: [i2s_in] AEL_MSG_CMD_RESUME,state:1

[15:28:02][D][esp-idf:000]: I (51058) AUDIO_PIPELINE: Pipeline started`

closed 2024-05-05 by pimp1310