gnumpi/esphome_audio #36: Playing a Mp3 File with a voice Command and a automation ====> ESP Crashes

pimp1310, 2024-05-04:
Hello,

When i try to play a mp3 file with a automation thats reacts to a spoken command, the esp crashes.

When i send it over Services it works well without crashes.

the command is the same, in the automation and when i use services

`service: media_player.play_media
target:
  entity_id: media_player.v_assist_kitchen_esp32nr168_s3_dev_media_player
data:
  media_content_type: music
  media_content_id: http://live-icy.gss.dr.dk/A/A12H.mp3
`

here is the Crash and reboot when triggered over the animation and the spoken command.

`23:26:08][D][esp_adf_pipeline:302]: State changed from STOPPING to UNINITIALIZED
[23:26:08][D][voice_assistant:439]: State changed from STOPPING_MICROPHONE to AWAITING_RESPONSE
[23:26:08][D][voice_assistant:563]: Event Type: 4
[23:26:08][D][voice_assistant:591]: Speech recognised as: " Schalte das Radio an."
[23:26:08][D][voice_assistant:563]: Event Type: 5
[23:26:08][D][voice_assistant:596]: Intent started
[23:26:08][D][media_player:059]: 's3-dev_media_player' - Setting
[23:26:08][D][media_player:066]:   Media URL: http://live-icy.gss.dr.dk/A/A12H.mp3
[23:26:08][D][adf_media_player:030]: Got control call in state 1
[23:26:08][D][esp_adf_pipeline:050]: Starting request, current state UNINITIALIZED
[23:26:08][D][esp-idf:000]: I (573009) MP3_DECODER: MP3 init

[23:26:08][D][esp-idf:000]: I (573021) I2S: DMA Malloc info, datalen=blocksize=512, dma_buf_count=4

[23:26:08][D][i2s_audio:072]: Installing driver : yes
[23:26:08][D][esp_adf_pipeline:358]: pipeline tag 0, http
[23:26:08][D][esp_adf_pipeline:358]: pipeline tag 1, decoder
[23:26:08][D][esp_adf_pipeline:358]: pipeline tag 2, i2s_out
[23:26:08][D][esp-idf:000]: I (573052) AUDIO_PIPELINE: link el->rb, el:0x3d8263a4, tag:http, rb:0x3d8268b0

[23:26:08][D][esp-idf:000]: I (573060) AUDIO_PIPELINE: link el->rb, el:0x3d82654c, tag:decoder, rb:0x3d8278f0

[23:26:08][D][esp_adf_pipeline:370]: Setting up event listener.
[23:26:08][D][esp_adf_pipeline:302]: State changed from UNINITIALIZED to PREPARING
[23:26:08][I][adf_media_player:135]: got new pipeline state: 1
[23:26:08][D][adf_i2s_out:127]: Set final i2s settings: 22050
[23:26:08][D][esp-idf:000]: I (573095) AUDIO_THREAD: The http task allocate stack on external memory

[23:26:08][D][esp-idf:000]: I (573101) AUDIO_ELEMENT: [http-0x3d8263a4] Element task created

[23:26:08][D][esp-idf:000]: I (573112) AUDIO_THREAD: The decoder task allocate stack on external memory

[23:26:08][D][esp-idf:000]: I (573121) AUDIO_ELEMENT: [decoder-0x3d82654c] Element task created

[23:26:09][D][esp-idf:000]: I (573123) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1

[23:26:09][D][esp-idf:000]: I (573132) AUDIO_ELEMENT: [decoder] AEL_MSG_CMD_RESUME,state:1

[23:26:09][D][esp_aud:000]: 
ERROR Fatal error: protocol.data_received() call failed.
protocol: <aioesphomeapi._frame_helper.noise.APINoiseFrameHelper object at 0x7f25b25d7c10>
transport: <_SelectorSocketTransport fd=6 read=polling write=<idle, bufsize=0>>
Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/selector_events.py", line 1009, in _read_ready__data_received
    self._protocol.data_received(data)
  File "aioesphomeapi/_frame_helper/noise.py", line 136, in aioesphomeapi._frame_helper.noise.APINoiseFrameHelper.data_received
  File "aioesphomeapi/_frame_helper/noise.py", line 163, in aioesphomeapi._frame_helper.noise.APINoiseFrameHelper.data_received
  File "aioesphomeapi/_frame_helper/noise.py", line 319, in aioesphomeapi._frame_helper.noise.APINoiseFrameHelper._handle_frame
  File "/usr/local/lib/python3.11/dist-packages/noise/state.py", line 74, in decrypt_with_ad
    plaintext = self.cipher.decrypt(self.k, self.n, ad, ciphertext)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/noise/backends/default/ciphers.py", line 13, in decrypt
    return self.cipher.decrypt(nonce=self.format_nonce(n), data=ciphertext, associated_data=ad)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "src/chacha20poly1305_reuseable/__init__.py", line 127, in chacha20poly1305_reuseable.ChaCha20Poly1305Reusable.decrypt
  File "src/chacha20poly1305_reuseable/__init__.py", line 147, in chacha20poly1305_reuseable.ChaCha20Poly1305Reusable.decrypt
  File "src/chacha20poly1305_reuseable/__init__.py", line 263, in chacha20poly1305_reuseable._decrypt_with_fixed_nonce_len
  File "src/chacha20poly1305_reuseable/__init__.py", line 273, in chacha20poly1305_reuseable._decrypt_data
cryptography.exceptions.InvalidTag
WARNING v-assist-kitchen-esp32nr168 @ 192.168.0.157: Connection error occurred: v-assist-kitchen-esp32nr168 @ 192.168.0.157: Invalid encryption key: received_name=v-assist-kitchen-esp32nr168
INFO Processing unexpected disconnect from ESPHome API for v-assist-kitchen-esp32nr168 @ 192.168.0.157
WARNING Disconnected from API
INFO Successfully connected to v-assist-kitchen-esp32nr168 @ 192.168.0.157 in 0.009s
INFO Successful handshake with v-assist-kitchen-esp32nr168 @ 192.168.0.157 in 0.118s
[23:26:09][D][esp-idf:000]: I (573327) HTTP_CLIENT: Body received in fetch header state, 0x3fccbd99, 1152

[23:26:09][D][esp-idf:000]: I (573331) HTTP_STREAM: total_bytes=0

[23:26:12][D][esp_adf_pipeline:400]: Called deinit_all
[23:26:14][D][esp-idf:000]: W (578286) AUDIO_ELEMENT: [http-0x3d8263a4] Element task destroy timeout[2000]

[23:26:16][D][esp-idf:000]: W (580288) AUDIO_ELEMENT: [decoder-0x3d82654c] Element task destroy timeout[2000]

[23:26:16][D][esp-idf:000]: W (580290) AUDIO_ELEMENT: [i2s_out] Element has not create when AUDIO_ELEMENT_TERMINATE

[23:26:16][D][esp-idf:000]: I (580292) AUDIO_PIPELINE: audio_pipeline_unlinked

WARNING v-assist-kitchen-esp32nr168 @ 192.168.0.157: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for v-assist-kitchen-esp32nr168 @ 192.168.0.157
WARNING Disconnected from API
INFO Successfully connected to v-assist-kitchen-esp32nr168 @ 192.168.0.157 in 0.009s
INFO Successful handshake with v-assist-kitchen-esp32nr168 @ 192.168.0.157 in 0.099s

`

gnumpi, 2024-05-07:
I also noticed some issues when trying to trigger a mp3 stream from a voice command. Actually there was an issue of simultaneously trying to play the tts response message and the requested audio stream. I re-wrote some of the code to prevent this. You can try if this also fixes your issue:
```yaml
- source:
      type: git
      url: https://github.com/gnumpi/esphome_audio
      ref: dev-next
    components: [ adf_pipeline, i2s_audio ]
```

pimp1310, 2024-05-07:
> I also noticed some issues when trying to trigger a mp3 stream from a voice command. Actually there was an issue of simultaneously trying to play the tts response message and the requested audio [...]

i give it a try, now when i start a mp3 stream over voice he says something  unreconizeable 

here is the log

`
[20:24:37][D][esp_adf_pipeline:444]: [ADFMicrophone] Pipeline changed from DESTROYING to UNINITIALIZED. (REQ: 1)
[20:24:37][D][esp_adf_pipeline:444]: [ADFMicrophone] Pipeline changed from UNINITIALIZED to UNINITIALIZED. (REQ: 1)
[20:24:38][D][voice_assistant:563]: Event Type: 4
[20:24:38][D][voice_assistant:591]: Speech recognised as: " Schalte das Radio an."
[20:24:38][D][voice_assistant:563]: Event Type: 5
[20:24:38][D][voice_assistant:596]: Intent started
[20:24:38][D][media_player:059]: 's3-dev_media_player' - Setting
[20:24:38][D][media_player:066]:   Media URL: http://live-icy.gss.dr.dk/A/A12H.mp3
[20:24:38][D][esp_audio_sources:058]: Set new uri: http://live-icy.gss.dr.dk/A/A12H.mp3
[20:24:38][D][adf_media_player:047]: Got control call in state IDLE
[20:24:38][D][esp_adf_pipeline:060]: Starting request, current state UNINITIALIZED
[20:24:38][D][esp_adf_pipeline:444]: [MediaPlayer] Pipeline changed from UNINITIALIZED to INITIALIZING. (REQ: 0)
[20:24:38][I][adf_media_player:167]: got new pipeline state: 1, while in MP state IDLE
[20:24:38][I][adf_media_player:226]: current mp state: IDLE
[20:24:38][I][adf_media_player:227]: anouncement: false
[20:24:38][I][adf_media_player:228]: play_intent: false
[20:24:38][I][adf_media_player:229]: current_uri_: yes
[20:24:38][D][voice_assistant:563]: Event Type: 6
[20:24:38][D][esp-idf:000]: I (263807) MP3_DECODER: MP3 init

[20:24:38][D][esp-idf:000]: I (263816) I2S: DMA Malloc info, datalen=blocksize=512, dma_buf_count=4

[20:24:38][D][i2s_audio:072]: Installing driver : yes
[20:24:38][D][esp_adf_pipeline:493]: pipeline tag 0, http
[20:24:38][D][esp_adf_pipeline:493]: pipeline tag 1, decoder
[20:24:38][D][esp_adf_pipeline:493]: pipeline tag 2, i2s_out
[20:24:38][D][esp-idf:000]: I (263844) AUDIO_PIPELINE: link el->rb, el:0x3d8263a4, tag:http, rb:0x3d832940

[20:24:38][D][esp-idf:000]: I (263846) AUDIO_PIPELINE: link el->rb, el:0x3d8325dc, tag:decoder, rb:0x3d833980

[20:24:38][D][esp_adf_pipeline:503]: Setting up event listener.
[20:24:38][D][esp_adf_pipeline:444]: [MediaPlayer] Pipeline changed from INITIALIZING to CREATED. (REQ: 0)
[20:24:38][I][adf_media_player:167]: got new pipeline state: 2, while in MP state IDLE
[20:24:38][I][adf_media_player:226]: current mp state: IDLE
[20:24:38][I][adf_media_player:227]: anouncement: false
[20:24:38][I][adf_media_player:228]: play_intent: false
[20:24:38][I][adf_media_player:229]: current_uri_: yes
[20:24:38][D][voice_assistant:563]: Event Type: 7
[20:24:38][D][voice_assistant:619]: Response: "Done"
[20:24:38][D][esp_adf_pipeline:444]: [MediaPlayer] Pipeline changed from CREATED to PREPARING. (REQ: 0)
[20:24:38][I][adf_media_player:167]: got new pipeline state: 3, while in MP state IDLE
[20:24:38][D][adf_i2s_out:127]: Set final i2s settings: 22050
[20:24:38][I][adf_media_player:226]: current mp state: PLAYING
[20:24:38][I][adf_media_player:227]: anouncement: false
[20:24:38][I][adf_media_player:228]: play_intent: false
[20:24:38][I][adf_media_player:229]: current_uri_: yes
[20:24:38][D][voice_assistant:563]: Event Type: 8
[20:24:38][D][voice_assistant:639]: Response URL: "http://192.168.0.73:8123/api/tts_proxy/e9b450d14bc2363d292c84f17cfad5cfbd58a458_de-de_1bd7a6bba9_tts.piper_2.mp3"
[20:24:38][D][voice_assistant:439]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[20:24:38][D][voice_assistant:445]: Desired state set to STREAMING_RESPONSE
[20:24:38][D][esp_audio_sources:063]: Prepare elements called (initial_call)!
[20:24:38][D][esp_audio_sources:097]: Use fixed settings: no
[20:24:38][D][esp_audio_sources:098]: Streamer status: 1
[20:24:38][D][esp_audio_sources:099]: decoder status: 1
[20:24:38][D][adf_audio_element:108]: Preparing [http]...
[20:24:38][D][esp-idf:000]: I (264018) AUDIO_THREAD: The http task allocate stack on external memory

[20:24:38][D][esp-idf:000]: I (264028) AUDIO_ELEMENT: [http-0x3d8263a4] Element task created

[20:24:38][D][adf_audio_element:108]: Preparing [decoder]...
[20:24:38][D][esp-idf:000]: I (264039) AUDIO_THREAD: The decoder task allocate stack on external memory

[20:24:39][D][esp-idf:000]: I (264049) AUDIO_ELEMENT: [decoder-0x3d8325dc] Element task created
[... 153 lines cut ...]
[20:24:43][D][esp_adf_pipeline:493]: pipeline tag 1, pcm_reader
[20:24:43][D][esp-idf:000]: I (268935) AUDIO_PIPELINE: link el->rb, el:0x3d8263e0, tag:i2s_in, rb:0x3d832754

[20:24:43][D][esp_adf_pipeline:503]: Setting up event listener.
[20:24:43][D][esp_adf_pipeline:444]: [ADFMicrophone] Pipeline changed from INITIALIZING to CREATED. (REQ: 0)
[20:24:43][D][esp_adf_pipeline:444]: [ADFMicrophone] Pipeline changed from CREATED to PREPARING. (REQ: 0)
[20:24:43][D][adf_audio_element:108]: Preparing [i2s_in]...
[20:24:43][D][esp-idf:000]: I (268968) AUDIO_ELEMENT: [i2s_in-0x3d8263e0] Element task created

[20:24:43][D][adf_audio_element:108]: Preparing [pcm_reader]...
[20:24:43][D][esp-idf:000]: I (268981) AUDIO_ELEMENT: [pcm_reader-0x3d83260c] Element task created

[20:24:43][D][esp_adf_pipeline:345]: wait for preparation, done
[20:24:43][D][esp_adf_pipeline:444]: [ADFMicrophone] Pipeline changed from PREPARING to STARTING. (REQ: 0)
[... cut ...]
[20:24:43][I][esp_adf_pipeline:135]: [ i2s_in ] status: 12
[20:24:44][D][esp_adf_pipeline:444]: [ADFMicrophone] Pipeline changed from STARTING to RUNNING. (REQ: 0)
[20:24:44][D][micro_wake_word:177]: State changed from STARTING_MICROPHONE to DETECTING_WAKE_WORD
[20:25:17][I][ota:117]: Boot seems successful, resetting boot loop counter.
[20:25:17][D][esp32.preferences:114]: Saving 1 preferences to flash...
[20:25:17][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed`

pimp1310, 2024-05-09:
@gnumpi 

have you seen my feedback to the new version in "dev-next" above?

gnumpi, 2024-05-09:
I am sorry, but playing music from voice commands is not supported until https://github.com/esphome/esphome/pull/6691 got merged.

pimp1310, 2024-05-12:
> I am sorry, but playing music from voice commands is not supported until [esphome/esphome#6691](https://github.com/esphome/esphome/pull/6691) got merged.

Okay, I'm not familiar with this, what are the chances of it being merged?

and when is something like this usually merged?

Thank You

gnumpi, 2024-05-12:
> > I am sorry, but playing music from voice commands is not supported until [esphome/esphome#6691](https://github.com/esphome/esphome/pull/6691) got merged.
> 
> Okay, I'm not familiar with this, what are the chances of it being merged?
> 
> and when is something like this usually merged?
> 
> Thank You

Actually, I don't know. There are quite a few PR for esphome, so the maintainers need to have their priorities and order of checking them. As this PR is mainly related to the ADF-pipeline which is only an external component for now, I guess it's not on their highest priority. 
But, in the meantime I filed another PR which doesn't include the API changes needed for sending the announcement flag from HA, and hence, it can be included as external source in your configuration:

Maybe you want to give it a try:
```yaml
external_components:
  - source:  github://pr#6728
    components: [ api, media_player, speaker, voice_assistant ]
  
  - source:
      type: git
      url: https://github.com/gnumpi/esphome_audio
      ref: dev-next

    components: [ adf_pipeline, i2s_audio ]

```

gnumpi, 2024-06-12:
Hey, thx for reporting your issue! Please, don't use the adf-mods anymore. All the necessary parts are already merged into the current ESPHome version. Can you confirm that your issue still exists with the latest dev-next commit and without the second external component in your config?

s00500, 2024-06-12:
Hi @gnumpi 

Yes it does, https://github.com/gnumpi/esphome_audio/commit/a8f66cbcb9ad96c71904225d0ee0e89641223247 is the latest dev-next commit

I only added the adf-mods after starting to play arround so the issue exists in anycase... unfortunatly it is also not super easy to get the logs from the onju, as it just looses them when crashing when using network logging 

But the issue is fairly consistent

gnumpi, 2024-06-12:
ok, let me try to reproduce the error with my hardware then ...

gnumpi, 2024-06-12:
unfortunately I couldn't reproduce your error, it is just working fine here. I don't have an Onju though, but I am running an ESP32-S3 with similar settings, and also in duplex mode.

s00500, 2024-06-17:
Ok, it always works fine on the first try, then somehow hangs up... seems like it is not crashing anymore with my current config though...

Here is my config and the logs of 3 attempts... notice I use encryption...

<details><summary>Configuration</summary><p>

```
# Wohnzimmer
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

ota:
  password: XXX

substitutions:
  name: "onju-voice-1"
  friendly_name: "Nabu 1"
  project_version: "1.0.0"
  device_description: "Onju Voice Satellite with ESPHome software and microWakeWord"
external_components:
  - source:
      type: git
      url: https://github.com/gnumpi/esphome_audio
      ref: d957129e0f37d622dd6e0efb2400bf2c9582a35c # main (high pitch issue)
      #ref: a8f66cbcb9ad96c71904225d0ee0e89641223247 #dev-next
    components: [adf_pipeline, i2s_audio]
  # - source:
  #     type: git
  #     url: https://github.com/gnumpi/esphome
  #     ref: 2044c7e4d4506b840dd2978d2fce5e455d304627 # adf-mods
  #   components: [api, media_player, speaker, voice_assistant]

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"
  comment: "${device_description}"
  name_add_mac_suffix: false
  project:
    name: tetele.onju_voice_satellite
    version: "${project_version}"
  min_version: 2024.3.0
  platformio_options:
    board_build.flash_mode: dio
    build_flags: "-DBOARD_HAS_PSRAM"
    board_build.arduino.memory_type: qio_opi
  on_boot:
    then:
      - light.turn_on:
          id: top_led
          effect: slow_pulse
          red: 100%
          green: 60%
          blue: 0%
      - wait_until:
          condition:
            wifi.connected:
      - light.turn_on:
          id: top_led
          effect: pulse
          red: 0%
          green: 100%
          blue: 0%
      - wait_until:
          condition:
            api.connected:
      - light.turn_on:
          id: top_led
          effect: none
[... 1026 lines cut ...]
  File "/opt/homebrew/lib/python3.11/site-packages/noise/backends/default/ciphers.py", line 13, in decrypt
    return self.cipher.decrypt(nonce=self.format_nonce(n), data=ciphertext, associated_data=ad)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "src/chacha20poly1305_reuseable/__init__.py", line 127, in chacha20poly1305_reuseable.ChaCha20Poly1305Reusable.decrypt
  File "src/chacha20poly1305_reuseable/__init__.py", line 147, in chacha20poly1305_reuseable.ChaCha20Poly1305Reusable.decrypt
  File "src/chacha20poly1305_reuseable/__init__.py", line 263, in chacha20poly1305_reuseable._decrypt_with_fixed_nonce_len
  File "src/chacha20poly1305_reuseable/__init__.py", line 273, in chacha20poly1305_reuseable._decrypt_data
cryptography.exceptions.InvalidTag
WARNING onju-voice-1 @ 192.168.1.181: Connection error occurred: onju-voice-1 @ 192.168.1.181: Invalid encryption key: received_name=onju-voice-1
INFO Processing unexpected disconnect from ESPHome API for onju-voice-1 @ 192.168.1.181
WARNING Disconnected from API
INFO Successfully connected to onju-voice-1 @ 192.168.1.181 in 0.192s
INFO Successful handshake with onju-voice-1 @ 192.168.1.181 in 0.055s
[23:11:07][D][esp-idf:000][decoder]: I (39498) MP3_DECODER: Closed

[23:11:07][D][esp-idf:000][resampler]: I (39593) AUDIO_ELEMENT: IN-[resampler] AEL_IO_DONE,-2

[23:11:07][D][esp-idf:000][i2s_out]: I (39641) AUDIO_ELEMENT: IN-[i2s_out] AEL_IO_DONE,-2

[23:11:07][D][esp_adf_pipeline:302]: State changed from STOPPING to STOPPED
[23:11:07][I][adf_media_player:135]: got new pipeline state: 5
[23:11:07][D][main:698]: DONE playing #####################
[23:11:07][D][main:223]: Started VA #####################
```
</p></details>

s00500, 2024-06-17:
On the second try it still plays the beep but then does not start the VA pipeline anymore

s00500, 2024-06-23:
upgraded to 2024.6 now... still feels like any media_player interaction I perform in the on_wakeword part of the config crashes it....

s00500, 2024-08-11:
Seems this is fine now after testing with dev-next and ESPHome 2024.7.3 !

Works good for me at least now :-)

DaAwesomeP, 2024-12-15:
I'm having the crash/reboot issue on 2024.11.3 on an M5Stack Core S3 SE when I play an MP3. I'm not having the same issue on the M5Stack Atom Echo (not using any custom components for that).

Here's my config (my esphome_m5stack_core_s3 is a fork of https://github.com/gnumpi/esphome_m5stack_core_s3):
```yaml
external_components:
  - source: local_components
    components: [touchscreen]
  - source:
      type: git
      url: https://github.com/DaAwesomeP/esphome_m5stack_core_s3
      ref: DaAwesomeP_touch_interrupt_and_backlight
    components: [ esp32_m5stack_core_s3, aw9523 ]
    refresh: 0s
  - source:
      type: git
      url: https://github.com/gnumpi/esphome_audio
      ref: dev-next
    components: [ adf_pipeline, i2s_audio ]

i2s_audio:
  - id: i2s_shared
    i2s_lrclk_pin: GPIO33
    i2s_bclk_pin: GPIO34
    i2s_mclk_pin: GPIO0
    access_mode: duplex

adf_pipeline:
  - platform: i2s_audio
    type: audio_out
    id: adf_i2s_out
    i2s_audio_id: i2s_shared
    i2s_dout_pin: GPIO13
    adf_alc: false
    dac:
      model: aw88298
      address: 0x36
      enable_pin:
        aw9523: aw9523_1
        port: 0
        pin: 2
        mode:
          output: true
    sample_rate: 16000
    bits_per_sample: 16bit
    fixed_settings: true

  - platform: i2s_audio
    type: audio_in
    id: adf_i2s_in
    i2s_audio_id: i2s_shared
    i2s_din_pin: GPIO14
    pdm: false
    adc:
      model: es7210
      address: 0x40
    bits_per_sample: 16bit
    fixed_settings: true

microphone:
  - platform: adf_pipeline
    id: adf_microphone
    keep_pipeline_alive: true
    pipeline:
      - adf_i2s_in
      - resampler
      - self

media_player:
  - platform: adf_pipeline
    id: adf_media_player
    name: None
    internal: false
    keep_pipeline_alive: true
    pipeline:
      - self
      - resampler
      - adf_i2s_out
 ```