gnumpi/esphome_audio #42: Resampler on input doesn't work in duplex mode

jherby2k, 2024-06-14:
Per https://github.com/gnumpi/esphome_audio/issues/17, in duplex mode we have to use 16000 Hz and skip the resampler if we want the mic to work. This means poor audio quality of course. This is affecting devices like the Onju (Google mini replacement PCB).

gnumpi, 2024-06-15:
Could you try with this branch if 44.1kHz is working with Onju?
please make sure to set both i2s pipeline elements to 44.1kHz and add the resampler to the mic pipeline.

```yaml
external_components:
  - source:
      type: git
      url: https://github.com/gnumpi/esphome_audio
      ref: fix_mic_resampler
    components: [ adf_pipeline, i2s_audio ]
```
please share your logs if it is not working.

dwyschka, 2024-06-16:
Still wont work :/ 

[logs_onju-voice-1_logs.txt](https://github.com/user-attachments/files/15858119/logs_onju-voice-1_logs.txt)

My ADF Config: 

`
i2s_audio:
  - id: i2s_shared
    i2s_lrclk_pin: GPIO13
    i2s_bclk_pin: GPIO18
    access_mode: duplex

adf_pipeline:
  - platform: i2s_audio
    type: audio_out
    id: adf_i2s_out
    i2s_audio_id: i2s_shared
    i2s_dout_pin: GPIO12
    sample_rate: 44100
    adf_alc: true
    bits_per_sample: 32bit
    fixed_settings: true
    channel: left

  - platform: i2s_audio
    type: audio_in
    id: adf_i2s_in
    i2s_audio_id: i2s_shared
    i2s_din_pin: GPIO17
    channel: left
    pdm: false
    sample_rate: 44100
    bits_per_sample: 32bit
    fixed_settings: true

microphone:
  - platform: adf_pipeline
    id: onju_microphone
    keep_pipeline_alive: true
    gain_log2: 3
    pipeline:
      - adf_i2s_in
      - resampler
      - self
 
media_player:
  - platform: adf_pipeline
    id: onju_out
    name: None
    internal: false
    keep_pipeline_alive: true
    pipeline:
      - self
      - resampler
      - adf_i2s_out
    on_state: 
      then:
        - lambda: |-
            static float old_volume = -1;
            float new_volume = id(onju_out).volume;
            if(abs(new_volume-old_volume) > 0.0001) {
              if(old_volume != -1) {
                id(show_volume)->execute();
              }
            }
            old_volume = new_volume;
`

Audio over MusicAssistant works fine.

gnumpi, 2024-06-16:
hmm, sorry to hear :( Could you share a bit more of you log file. So that the full life-cycle of the mic is shown?

dreimer1986, 2024-06-17:
I can provide you a log, but I am not sure if my "fix" is right. https://github.com/tetele/onju-voice-satellite/issues/52#issuecomment-2171476791

Github once more casues MADNESS in line 154 where by magic my "audio-in" written there suddently becomes audioIn after a bit. O_o I can build a few times and then Github magically changed it back. So take this one with a bit of salt.

gnumpi, 2024-06-17:
thx for sharing your logs!
I just committed a new fix to the same branch "fix_mic_resampler"
let's see ;)

gnumpi, 2024-06-17:
after a few more iterations, thx @dwyschka for live testing, the "fix_mic_resampler" branch seems to work now. 
I will test if it stills works with all my test devices and then going to merge to dev-next, and hopefully soon also to main.

closed 2024-07-05 by gnumpi

tetele, 2024-07-12:
I start getting these warnings when I change the sample rate to 44.1kHz

```
[17:39:40][W][micro_wake_word:103]: Not enough free bytes in ring buffer to store incoming audio data (free bytes=896, incoming bytes=1024). Resetting the ring buffer. Wake word detection accuracy will be reduced.
[17:39:40][W][micro_wake_word:103]: Not enough free bytes in ring buffer to store incoming audio data (free bytes=896, incoming bytes=1024). Resetting the ring buffer. Wake word detection accuracy will be reduced.
[17:39:41][W][micro_wake_word:103]: Not enough free bytes in ring buffer to store incoming audio data (free bytes=896, incoming bytes=1024). Resetting the ring buffer. Wake word detection accuracy will be reduced.
[17:39:41][W][micro_wake_word:103]: Not enough free bytes in ring buffer to store incoming audio data (free bytes=896, incoming bytes=1024). Resetting the ring buffer. Wake word detection accuracy will be reduced.
[17:39:42][W][micro_wake_word:103]: Not enough free bytes in ring buffer to store incoming audio data (free bytes=640, incoming bytes=1024). Resetting the ring buffer. Wake word detection accuracy will be reduced.
```

I guess it's because the ring buffer size was halved in this PR https://github.com/esphome/esphome/pull/6830 but the sample rate there is hard-coded to 16kHz