gnumpi/esphome_audio #45: TTS response from chatgpt stops playback at around 15seconds

BigBobbas, 2024-06-27:
Hi Mischa , I hope you are well. 
I have the project with the S3 box 3 which is going well but i have had an issue logged where TTS responses that are circa 15secs the audio stops. I've not set this up yet to test and though I would ask if you knew of any reason why this would be happening.
The yaml for the device is here https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/blob/main/s3b.yaml

The issue raised is here https://github.com/BigBobbas/ESP32-S3-Box3-Custom-ESPHome/issues/27 which includes logs from the user.

Any ideas or pointers would be greatly appreciated.

Cheers 
Rob

formatBCE, 2024-07-06:
There's some check failed in PR, but looks like it's just new mandatory OTA platform field missing in test YAML. :)

closed 2024-07-10 by gnumpi

formatBCE, 2024-07-10:
@gnumpi i updated the code today, tried to play long announcement, and still it's cutting TTS stream. I see that player is stopping itself, with no apparent errors in logs. I checked the mp3 URL from logs, and there's like 1m30s of audio, containing the whole message - but logs saying "no more data" and player is stopping.
Is it memory issue? I can provide full log on demand.