gnumpi/esphome_audio #46: Media player repeats last played sound after each TTS response, if wasn't stopped explicitly

formatBCE, 2024-06-28:
IDK if it's something worth fixing, but as of now, on config with newly introduced timers from ESPHome, when i play "ding" for timer on media_player, and DON'T explicitly call "stop" on media_player after, each TTS response after that is accompanied with "ding" after. 
It looks like new status "ANNOUNCING" calls "PLAY" after announcement, regardless of previous player state (i assume it's paused there, probably, after timer's "ding"). 

Same thing happens, if i play anything on player from HA, let it play to the end, and then call some TTS - it will play< whatever was there before, after TTS. But doesn't happen, if i click "stop" button in HA while player is still playing.

BigBobbas, 2024-06-30:
I added the below to `media_payer: `component as a workaround and this appears to work fine with this added as I was having the same behaviour.
```yaml
on_idle:
  media_player.stop 
  ```

formatBCE, 2024-06-30:
> I added the below to `media_payer: `component as a workaround and this appears to work fine with this added as I was having the same behaviour.
> ```yaml
>     on_idle:
>       - media_player.stop 
>   ```

That's elegant. :)

gnumpi, 2024-07-03:
Thanks for pointing at this, actually it should only call play if it was in playing state when the announcement started. The hint that it plays tracks that has already been finished playing, might be very helpful. Probably, I forgot reseting the state somewhere...

gnumpi, 2024-07-03:
I might have found the problem, can you check if this fix (mp_reset_current_track_on_finish branch) helps?
```yaml
external_components:
  - source:
      type: git
      url: https://github.com/gnumpi/esphome_audio
      ref: mp_reset_current_track_on_finish
    components: [ adf_pipeline, i2s_audio ]
    refresh: 0s
```

formatBCE, 2024-07-03:
> I might have found the problem, can you check if this fix (mp_reset_current_track_on_finish branch) helps?
>
> ```yaml
> external_components:
>   - source:
>       type: git
>       url: [...]

Hello! Sorry for delay, i'm in PST time zone.. :)
Yes, it works on this branch. I removed `stop` line, and still it doesn't play anything after TTS.

gnumpi, 2024-07-03:
perfect, then I will merge it into the dev-next branch

gnumpi, 2024-07-05:
done

closed 2024-07-05 by gnumpi