gnumpi/esphome_audio #47: dev-next branch Music Assistant group player issue

darki73, 2024-07-01:
Hey, first of all, thank you for the great work you have done!

## Issue
While using a single speaker to play anything from Music Assistant, stream goes file, everything works.
The only missing piece to make formats other than `mp3` to work was adding `codec: auto` to the media player configuration block.

However, upon the creating a group of speakers, any time i try to play the audio, it results in the following errors on the device (each of them)

<details>
<summary>Log output from the device</summary>

```
[20:59:29][W][esp-idf:000]: I (356675) tsens: Config temperature range [-10°C ~ 80°C], error < 1°C

[20:59:51][W][esp-idf:000]: I (378667) ESP_DECODER: esp_decoder_init, stack size is 30720

[20:59:51][W][esp-idf:000]: I (378671) I2S: DMA Malloc info, datalen=blocksize=2048, dma_buf_count=4

[20:59:51][W][esp-idf:000]: I (378681) AUDIO_PIPELINE: link el->rb, el:0x3d8188d4, tag:http, rb:0x3d81909c

[20:59:51][W][esp-idf:000]: I (378691) AUDIO_PIPELINE: link el->rb, el:0x3d818bcc, tag:decoder, rb:0x3d8590dc

[20:59:51][W][esp-idf:000]: I (378693) AUDIO_PIPELINE: link el->rb, el:0x3d818d68, tag:resampler, rb:0x3d89911c

[20:59:51][W][esp-idf:000]: I (378723) AUDIO_THREAD: The http task allocate stack on external memory

[20:59:51][W][esp-idf:000]: I (378726) AUDIO_ELEMENT: [http-0x3d8188d4] Element task created

[20:59:51][W][esp-idf:000]: I (378728) AUDIO_THREAD: The decoder task allocate stack on external memory

[20:59:51][W][esp-idf:000]: I (378732) AUDIO_ELEMENT: [decoder-0x3d818bcc] Element task created

[20:59:51][W][esp-idf:000]: I (378747) AUDIO_THREAD: The resampler task allocate stack on external memory

[20:59:51][W][esp-idf:000]: I (378750) AUDIO_ELEMENT: [resampler-0x3d818d68] Element task created

[20:59:51][W][esp-idf:000]: I (378764) AUDIO_ELEMENT: [i2s_out-0x3d818f24] Element task created

[20:59:51][W][esp-idf:000][decoder]: I (378804) AUDIO_ELEMENT: [decoder] AEL_MSG_CMD_RESUME,state:1

[20:59:51][W][esp-idf:000][decoder]: I (378808) ESP_DECODER: Ready to do audio type check, pos:0, (line 104)

[20:59:51][W][esp-idf:000][http]: I (378813) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1

[20:59:51][W][esp-idf:000][http]: I (378841) HTTP_CLIENT: Body received in fetch header state, 0x3fccf247, 47

[20:59:51][W][esp-idf:000][http]: I (378846) HTTP_STREAM: total_bytes=47

[20:59:51][W][esp-idf:000][http]: E (378849) HTTP_STREAM: Invalid HTTP stream, status code = 404

[20:59:51][W][esp-idf:000][http]: E (378854) AUDIO_ELEMENT: [http] AEL_STATUS_ERROR_OPEN,-1

[20:59:51][W][esp-idf:000][http]: W (378859) AUDIO_ELEMENT: [http] audio_element_on_cmd_error,7

[20:59:51][W][esp-idf:000][decoder]: W (378866) AUDIO_ELEMENT: IN-[decoder] AEL_IO_ABORT

[20:59:51][W][esp-idf:000][decoder]: E (378871) AUDIO_FORMAT: failed to get head buf data(line 175)

[20:59:51][W][esp-idf:000][decoder]: E (378878) ESP_DECODER: Audio type detect error (line 147)

[20:59:51][W][esp-idf:000][decoder]: W (378883) AUDIO_ELEMENT: [decoder] AEL_IO_ABORT, -3

[20:59:51][E][esp_adf_pipeline:223]: HTTPStreamReader got in error state while PREPARING. Stopping pipeline!
[20:59:51][E][esp_adf_pipeline:246]: Timeout while PREPARING. Stopping pipeline!
[20:59:51][W][esp-idf:000]: E (378907) AUDIO_ELEMENT: [http] Element already stopped

[20:59:51][E][esp_adf_pipeline:223]: HTTPStreamReader got in error state while STOPPING. Stopping pipeline!
[20:59:51][E][esp_adf_pipeline:246]: Timeout while STOPPING. Stopping pipeline!
[20:59:51][W][esp-idf:000]: I (378926) AUDIO_PIPELINE: audio_pipeline_unlinked

[... 129 lines cut ...]
    pipeline:
      - adf_i2s_in
      - self

media_player:
  - platform: adf_pipeline
    id: i2s_player
    name: "${device_friendly_name} Media Player"
    keep_pipeline_alive: false
    codec: auto
    internal: false
    announcement_audio:
      sample_rate: 16000
      bits_per_sample: 16
      num_channels: 1
    pipeline:
      - self
      - resampler
      - adf_i2s_out
```

## Additional Information
I am fully aware that code from the `dev-next` is not intended to be used on a "daily basis", so i can't really say that there is an issue with the project itself.

Just want to throw out there the issue i had, better to say it early than to refactor a lot of code later.

gnumpi, 2024-07-03:
I haven't looked into the speaker groups capability of MA yet, it seems MA creates a url in that case that is not recognisable by the http client. Can you copy paste the stream URL from the log and try playing it in your browser?

darki73, 2024-07-03:
@gnumpi Hey, thanks for the reply. Just give me a couple  of minutes to recompile the firmware with logging enabled and i will update this message once i will have results.

gnumpi, 2024-07-03:
> And no, i am unable to play the audio in the browser as it seems to expect something else rather than just URL from me. With the `arduino` and with no `esphome_audio` it works just fine with the same URL on the device, but yet again, from the browser i am not able to play the stream.

I was not aware that the arduino media player does play flac streams.

Is it possible that the url is only valid for playing once? Because the esphome_audio media player streams the url silently for a bit to detect the necessary audio settings and then finally plays the url again in a second step after all settings are made.

darki73, 2024-07-03:
I seriously doubt that the stream is in `flac`, the original file is an `m4a` (youtube music as a source) or `ogg` (spotify), my guess the headers which device receives during the connection phase actually specify which format it is supposed to use.

Reasons why i don't actually believe that stream is in `flac`:
1. There is no `ffmpeg` installed on my system
2. The CPU usage during the playback is not rising (it is a 6800H system which runs Proxmox and Home Assistant as a VM).

On another note, when trying to play the audio to a group which is using `esp-idf`, logs of the MA show the following (i know that this might be not related, and an error from the completely different project, but still):
```
2024-07-03 16:24:43.006 INFO (MainThread) [music_assistant.streams] Start Queue Flow stream for Queue Bedroom - crossfade: False
2024-07-03 16:24:43.536 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 1013, in _read_ready__data_received
    data = self._sock.recv(self.max_size)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ConnectionResetError: [Errno 104] Connection reset by peer

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 452, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_app.py", line 543, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/music_assistant/server/helpers/webserver.py", line 136, in _handle_catch_all
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/music_assistant/server/providers/ugp/__init__.py", line 392, in _serve_ugp_stream
    await resp.write(chunk)
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_response.py", line 517, in write
    await self._payload_writer.write(data)
  File "/usr/local/lib/python3.12/site-packages/aiohttp/http_writer.py", line 119, in write
    await self.drain()
  File "/usr/local/lib/python3.12/site-packages/aiohttp/http_writer.py", line 171, in drain
    await self._protocol._drain_helper()
  File "/usr/local/lib/python3.12/site-packages/aiohttp/base_protocol.py", line 95, in _drain_helper
    await asyncio.shield(waiter)
ConnectionError: Connection lost
2024-07-03 16:24:43.540 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 1013, in _read_ready__data_received
    data = self._sock.recv(self.max_size)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ConnectionResetError: [Errno 104] Connection reset by peer

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 452, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_app.py", line 543, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/music_assistant/server/helpers/webserver.py", line 136, in _handle_catch_all
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/music_assistant/server/providers/ugp/__init__.py", line 392, in _serve_ugp_stream
    await resp.write(chunk)
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_response.py", line 517, in write
    await self._payload_writer.write(data)
  File "/usr/local/lib/python3.12/site-packages/aiohttp/http_writer.py", line 119, in write
    await self.drain()
  File "/usr/local/lib/python3.12/site-packages/aiohttp/http_writer.py", line 171, in drain
    await self._protocol._drain_helper()
  File "/usr/local/lib/python3.12/site-packages/aiohttp/base_protocol.py", line 95, in _drain_helper
    await asyncio.shield(waiter)
ConnectionError: Connection lost
```

gnumpi, 2024-07-03:
Ok, thanks for all the useful information. I will need to try to reproduce the error on my setup. It will probably take a few days though.

darki73, 2024-07-03:
Sure, no problem, as i said, i am aware that this is a dev-next branch which should not be used without expecting some issues )

formatBCE, 2024-07-03:
@darki73 side question: do you connect the player to MA over HA?

darki73, 2024-07-03:
@formatBCE hey, yes, players flashed from esphome ui in homeassistant, then added to homeassistant under esphome integration, then enabled in the Home Asisstant players in MA.

Standalone players work just fine with `esp-idf` firmware, but not when grouped, with `arduino` I am able to play same media on all 4 (still assembling rest) speakers with no issue while all of them in a group.

formatBCE, 2024-07-03:
@darki73 that's interesting. I tried it once more, same result - doesn't play.. I tried with YouTube, Spotify and Jellyfin music providers - everything gets translated into MA flac HTTP url, and then I see 0 bytes stream in esp logs..

formatBCE, 2024-07-03:
@gnumpi it's even more interesting, that i actually can play music directly from Home Assistant (one by one, apparently, but it shouldn't matter), but not from Music Assistant... The URL from HA is mp3, while from MA it's always flac, regardless of the source library. Could this be the issue?

darki73, 2024-07-04:
Here are some more findings:
1. https://github.com/music-assistant/server/blob/main/music_assistant/server/providers/ugp/__init__.py#L215
2. https://github.com/music-assistant/server/blob/main/music_assistant/server/providers/ugp/__init__.py#L337
3. https://github.com/music-assistant/server/blob/main/music_assistant/server/providers/ugp/__init__.py#L298

They do indeed just straight up slap the `.flac` to the URL, which, by my guess, might not be handled properly in `esp-adf`, and by proper i mean framework sees the extension, and just goes with it.

Thing is, i have tried the YouTube streams with `m4a` and Spotify streams with `ogg` when playing solo, and MA is indeed passes the extension of the stream to ESP32, however, when using a group player (UGP -> Universal Group Player), according to the 3 links i provided earlier, it straight up just adds `.flac` and goes with it.

gnumpi, 2024-07-04:
In my understanding there are two different issues here, the first one is that MA ignores the enforce mp3 setting since its last update, as it was already reported here: https://github.com/music-assistant/hass-music-assistant/issues/2508. 
Because of this, the audio is always send as flac and the 'codec: auto' is needed to be able to play flac with the adf_mediaplayer. 
The second issue is probably, that the audio stream can't be set up correctly on MA side after it got aborted by the client. 
In order to test if this is the case, I could add a `music_assistant_audio` settings (like `the announcement_audio`) to the media player which defines in which format audio is send from MA and when the MP detects a stream coming from MA it uses this settings instead of auto detecting the format. This would prevent the need of aborting the stream and restarting it after the audio settings got detected. 
This is actually already on my todo list, but didn't get to it yet...  

@darki73 how do you run MA? Is it possible that MA has installed its own ffmpeg? Because I assume that MA depends on ffmepg and it always recodes the stream into either flac or mp3.

darki73, 2024-07-04:
@gnumpi MA is installed as an addon from the Home Assistant (i run Home Assistant OS).

The reason i doubt that it makes the UGP decode FLAC as i am not noticing any CPU usage increase during the playback (assuming it resamples the m4a/ogg to flac), and the machine is using mobile Ryzen 6800H CPU, while only 4 cores dedicated to a VM, so there should be at least a small spike during the playback initialization, but there is none. Also, they are making 2 URLs for the UGP stream, one `mp3` and one `flac`, but that means that they have to do double the amount of work to actually provide two different streams with two different formats.

Funny thing is, if you click "enough" on the play/pause button, the sound will actually comes live and i can either hear it from first / seconds / both of the speakers, BUT, from the logs, it is still the `flac` URL

formatBCE, 2024-07-04:
@darki73 lossless FLAC is a bit different from MP3/AAC/OGG, converted to FLAC, where all lossy bytes will be replaced with 0. Also it might be that MA is "presenting" FLAC stream, and actually is showing MP3 data inside - but headers to stream are FLAC, so MP3 codec can't read the stream. This might be a stretch, but it aligns with what we see on practice.

darki73, 2024-07-04:
@formatBCE I have a board of my custom design with all the necessary components for playback and voltage management being places as close to the ESP as possible. With the off the self boards I had many issues with sound quality both in recording and outputting.

gnumpi, 2024-07-04:
@darki73 just for clarification, the HA addon is based on the server docker image, hence ffmpeg is installed and runs inside the MA docker container on your system:

https://github.com/music-assistant/server/blob/496b99d0b3681f3b65744b0f9ca7ba718015efa4/Dockerfile#L64

formatBCE, 2024-07-08:
BTW, i also found that wit codec: auto, while actually playing the song, media player somehow reports back that it skipped the song to MA, and in MA interface it starts showing next song in queue, although player actually is playing the same first song. If i pause/unpause or seek with seekbar from MA, it shows old song for couple seconds, then behaviour repeats. But If i give announcement during this, then after announcement player starts playing NEW song, not the song it stopped on.
@gnumpi i guess it might be cache issue? 
Anyways, thank you for your work, it's the best implementation to the MP on ESPHome i've seen so far. If you need my help in testing, i'm happy to help. I wish i could dig into this deeper, but it's above my C++ skills. :)

gnumpi, 2024-07-10:
Yeah unfortunately the MA support for native HA media players is not very good in the moment. They prefer providers that speak directly from MA to the devices instead of taking the route through HA. Unfortunately I don't have the time at the moment to also implement a direct route to MA.

formatBCE, 2024-07-10:
> Yeah unfortunately the MA support for native HA media players is not very good in the moment. They prefer providers that speak directly from MA to the devices instead of taking the route through HA. Unfortunately I don't have the time at the moment to also implement a direct route to MA.

You mean the provider for ESPHome? That would be a longshot! :)

darki73, 2024-07-21:
Small update, with https://github.com/gnumpi/esphome_audio/commit/d628b9734b54794e57739b53ec18117f772ee79e even the real FLAC can be played with no issues. ESP32-S3 is such a powerhouse!