gnumpi/esphome_audio #49: alc_max above 0.4 causes distortion.

jonahnm, 2024-07-01:
As far as I can tell, this is not caused by my dac, amp or speaker as it happens even if I lower the gain on the amp itself.

DAC/AMP: MAX98357A

darki73, 2024-07-08:
From my observation, any AMP which is not soldered to the board will have a distortion.
You may want to try it out with the very short cables going from pins to the AMP.

On my device, i have the AMP placed on the circuit board itself, so it has direct connection to the ESP32, and the sound is crystal clear even with `alc_max` set to 0.5 (which from my understanding actually limits the max volume, and indeed would cause a distortion if there is a signal degradation when set too high).

jonahnm, 2024-07-08:
I could try to directly solder it to the board! Thanks for the tip!

jonahnm, 2024-07-09:
> From my observation, any AMP which is not soldered to the board will have a distortion. You may want to try it out with the very short cables going from pins to the AMP.
>
> On my device, i have the [...]

I've soldered it directly to the ESP32, same issue.

darki73, 2024-07-09:
Mine is on the PCB itself, so it uses traces to connect to ESP.

One more thing to try, you can set the gain using the GAIN pad, thing is, if it left hanging, it will increase gain by default to +9dB.

In the datasheet for the AMP you can see what you need to do to achieve different values of GAIN, from +3dB to +15dB.

On +15dB and 100% volume with alc set to 0.5 even with my setup I can hear distortion.

jonahnm, 2024-07-10:
Yes, but I'm not sure where the distortion is coming from, the AMP, or the ESP.

gnumpi, 2024-07-10:
Have you tried a setup without the adf_pipeline? E.g. one of the officially provided ESPHome firmwares? Do you see the same issue then?

jonahnm, 2024-07-10:
> Have you tried a setup without the adf_pipeline? E.g. one of the officially provided ESPHome firmwares? Do you see the same issue then?

The official ones don't work at all for me.

jonahnm, 2024-07-10:
(As in streaming on the official ones are broken)

jonahnm, 2024-07-11:
> From my observation, any AMP which is not soldered to the board will have a distortion. You may want to try it out with the very short cables going from pins to the AMP.
>
> On my device, i have the [...]

I will note that after soldering directly to the ESP32, alc_max of 0.5 sounds quite clear now.

jonahnm, 2024-07-11:
There is still some distortion on louder music though.

jonahnm, 2024-07-11:
I might end up buying a dedicated DAC instead of the combo and using the dedicated amps I have lying around (PAM8403) instead.

darki73, 2024-07-12:
@jonahnm PAM8403 in my experience (i have 20 of them ready to go to the trash bin) is a pretty bad AMP. If you are willing to do things right (read the datasheet for it and then filter the signal) and want to have a good sound, you kinda have to go MAX98357.

I have tried out about 5 or 6 PAMs and none of them were able to produce the sound to my liking.

Here is the example i can provide you with recorded on the Galaxy S24 Ultra from 2 meters away, device pointed at me, (sorry for applied volume normalization, AI stuff is killing me on this device) with volume being set to 100% on my speaker (also, had to use a dodgy online converter to make a WEBM out of M4A file, so there might be some sound quality degradation).

[Voice 240713_033530.webm](https://github.com/user-attachments/assets/746ffe0a-a9d5-4224-830e-7c2786056ddc)

jonahnm, 2024-07-15:
I know right, lol, anyways I'm trying to make my own implementation for everything using PWM audio with a low pass filter rather than a dedicated AMP. I also bought ESP32-S3s as they have more RAM, just in case the clipping was a decoder issue.