gnumpi/esphome_audio #50: Very fast TTS on [some] Piper announcements

formatBCE, 2024-07-03:
First of all - push me in required direction if needed.

The problem: TTS announcements for Piper (especially, seems like, for some voices there) are been played on speed like 2x faster than they should. 
For HA cloud and Google TTS it's all ok.
I use resampler in my media_player pipeline, if it's important. 
From logs, i see different bitrates for all TTS sources, if it's important. 
Will provide any debug info. :)

Thank you in advance!

gnumpi, 2024-07-03:
Normally, the audio settings gets automatically detected. Please check your config if the media player has an announcement_audio settings, if so these values are used instead of auto detection.

formatBCE, 2024-07-03:
You're right all the way through... I copied ESP32-S3-BOX config from one clever guy, and that setting slipped through completely unnoticed. Sorry for disturbance.

closed 2024-07-03 by formatBCE