gnumpi/esphome_audio #56: Allow media_player to play back local (PCM) audio

jhbruhn, 2024-07-19:
For (voice assistant related) functionalities like timer-alarms and awake-sounds after the wake-word has been detected, it would be great if the media_player component could play a locally stored PCM sound. For example in the esphome S3 Box firmware, [this is used to play an alarm sound](https://github.com/esphome/firmware/blob/a7db5f3dde6be30ebfd0ad2543a772a546433a45/wake-word-voice-assistant/esp32-s3-box-3.yaml#L284) with the speaker component. But when using a speaker, you don't get a media_player usable from HA.

The workaround would be to host these sounds on a local HTTP Server and play back from there, but that isn't low latency and sounds like an unnecessary complication to me.

It of course would be great to integrate this into ESPHomes `media_player` natively, but that [feature-request](https://github.com/esphome/feature-requests/issues/2704) is still open.

Alternative Idea:

Is it feasible to provide a generic media_player, which has just an output speaker-component? But then of course the volume etc. is ignored when accessing the speaker directly.

jhbruhn, 2024-07-25:
It seems like NabuCasa is working on a (XMOS specific?) Media Player implementation which will allow playback of local files already: https://github.com/esphome/voice-kit/tree/dev Maybe we should wait to see whether this aids in any way.

jnuha, 2025-05-28:
+ 1, i want play file local