gnumpi/esphome_audio #64: MediaPlayer in voice_assistant

KVinS, 2024-09-06:
Hello, I tried to use this component in voice_assistant, but get errors:

```
03:34:03 | [D] | [esp-idf:000][1;31m[http][0;36m | [0;32mI (67214) HTTP_STREAM: total_bytes=17146
03:34:03 | [I] | [HTTPStreamReader:129] | [ * ] Receive music info from mp3 decoder, sample_rates=22050, bits=16, ch=1
03:34:03 | [D] | [adf_i2s_out:114] | update i2s clk settings: rate:22050 bits:16 ch:1
03:34:03 | [D] | [esp-idf:000] | [0;32mI (67234) I2S: DMA Malloc info, datalen=blocksize=1024, dma_buf_count=4
03:34:03 | [D] | [adf_i2s_out:127] | Set final i2s settings: 22050
03:34:03 | [D] | [esp-idf:000][1;31m[decoder][0;36m | [0;33mW (67238) AUDIO_ELEMENT: OUT-[decoder] AEL_IO_ABORT
03:34:03 | [D] | [esp-idf:000][1;31m[decoder][0;36m | [0;33mW (67242) MP3_DECODER: output aborted -3
03:34:03 | [D] | [esp-idf:000][1;31m[decoder][0;36m | [0;32mI (67245) MP3_DECODER: Closed
03:34:03 | [D] | [esp-idf:000][1;31m[http][0;36m | [0;33mW (67250) HTTP_STREAM: No output due to stopping
03:34:03 | [D] | [esp_adf_pipeline:302] | State changed from PREPARING to STARTING
03:34:03 | [I] | [adf_media_player:135] | got new pipeline state: 2
03:34:03 | [D] | [adf_i2s_out:127] | Set final i2s settings: 22050
03:34:03 | [D] | [esp-idf:000] | [0;32mI (67260) AUDIO_ELEMENT: [i2s_out-0x3de0557c] Element task created
03:34:03 | [D] | [esp-idf:000] | [0;32mI (67261) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:2099211 Bytes, Inter:48280 Bytes, Dram:48280 B
03:34:03 | [D] | [esp-idf:000][1;31m[http][0;36m | [0;32mI (67262) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1
03:34:03 | [D] | [esp-idf:000][1;31m[decoder][0;36m | [0;32mI (67264) AUDIO_ELEMENT: [decoder] AEL_MSG_CMD_RESUME,state:1
03:34:03 | [D] | [esp-idf:000][1;31m[i2s_out][0;36m | [0;32mI (67265) AUDIO_ELEMENT: [i2s_out] AEL_MSG_CMD_RESUME,state:1
03:34:03 | [D] | [esp-idf:000][1;31m[i2s_out][0;36m | [0;32mI (67266) I2S_STREAM: AUDIO_STREAM_WRITER
03:34:03 | [I] | [esp_adf_pipeline:214] | [ http ] status: 14
03:34:03 | [I] | [esp_adf_pipeline:214] | [ i2s_out ] status: 12
03:34:03 | [D] | [esp_adf_pipeline:131] | Check element [http] status, 2
03:34:03 | [D] | [esp-idf:000][1;31m[http][0;36m | [0;32mI (67345) HTTP_CLIENT: Body received in fetch header state, 0x3fcde407, 1841
03:34:03 | [D] | [esp-idf:000][1;31m[http][0;36m | [0;32mI (67349) HTTP_STREAM: total_bytes=17146
03:34:03 | [I] | [esp_adf_pipeline:214] | [ http ] status: 12
03:34:03 | [D] | [esp_adf_pipeline:131] | Check element [http] status, 3
03:34:03 | [D] | [esp_adf_pipeline:131] | Check element [decoder] status, 2
03:34:03 | [D] | [esp_adf_pipeline:131] | Check element [i2s_out] status, 3
03:34:03 | [D] | [esp_adf_pipeline:302] | State changed from STARTING to RUNNING
03:34:03 | [I] | [adf_media_player:135] | got new pipeline state: 3
03:34:03 | [D] | [adf_i2s_out:127] | Set final i2s settings: 22050
03:34:03 | [I] | [esp_adf_pipeline:214] | [ decoder ] status: 12
03:34:03 | [I] | [HTTPStreamReader:129] | [ * ] Receive music info from mp3 decoder, sample_rates=22050, bits=16, ch=1
03:34:03 | [D] | [adf_i2s_out:127] | Set final i2s settings: 22050
03:34:04 | [D] | [esp-idf:000][1;31m[http][0;36m | [0;33mW (68145) HTTP_STREAM: No more data,errno:0, total_bytes:17146, rlen = 0
03:34:04 | [D] | [esp-idf:000][1;31m[http][0;36m | [0;32mI (68150) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0
03:34:04 | [I] | [esp_adf_pipeline:214] | [ http ] status: 15
03:34:04 | [D] | [esp_adf_pipeline:302] | State changed from RUNNING to STOPPING
03:34:04 | [I] | [adf_media_player:135] | got new pipeline state: 4
03:34:04 | [D] | [esp-idf:000][1;31m[decoder][0;36m | [0;32mI (68562) AUDIO_ELEMENT: IN-[decoder] AEL_IO_DONE,-2
03:34:05 | [D] | [esp-idf:000][1;31m[decoder][0;36m | [0;32mI (69002) MP3_DECODER: Closed
03:34:05 | [D] | [esp-idf:000][1;31m[i2s_out][0;36m | [0;32mI (69117) AUDIO_ELEMENT: IN-[i2s_out] AEL_IO_DONE,-2
03:34:05 | [D] | [esp_adf_pipeline:400] | Called deinit_all
03:34:05 | [D] | [esp-idf:000] | [0;32mI (69305) AUDIO_PIPELINE: audio_pipeline_unlinked
03:34:05 | [D] | [esp-idf:000] | [0;33mW (69306) AUDIO_ELEMENT: [http] Element has not create when AUDIO_ELEMENT_TERMINATE
03:34:05 | [D] | [esp-idf:000] | [0;33mW (69308) AUDIO_ELEMENT: [decoder] Element has not create when AUDIO_ELEMENT_TERMINATE
03:34:05 | [D] | [esp-idf:000] | [0;33mW (69309) AUDIO_ELEMENT: [i2s_out] Element has not create when AUDIO_ELEMENT_TERMINATE
03:34:05 | [D] | [esp-idf:000] | [0;32mI (69310) I2S: DMA queue destroyed
03:34:05 | [D] | [esp_adf_pipeline:302] | State changed from STOPPING to UNINITIALIZED

```

My code:

```
external_components:
  - source:
      type: git
      url: https://github.com/gnumpi/esphome_audio
      ref: main
      #type: local
      #path: /Users/siekmann/Privat/Projects/espHome/esphome_audio/esphome/components
    components: [ adf_pipeline, i2s_audio ]

substitutions:
  name: esphome-smart-black-speaker
  friendly_name: Black speaker

[... 185 lines cut ...]
              red: 0%
              green: 50%
              blue: 50%
              brightness: 30%
          - micro_wake_word.start:
          - light.turn_off: led_ww

light:
  - platform: esp32_rmt_led_strip
    id: led_ww
    rgb_order: GRB
    pin: GPIO21
    num_leds: 1
    rmt_channel: 0
    chipset: ws2812
    name: "on board light"
    effects:
      - pulse:
      - pulse:
          name: "Fast Pulse"
          transition_length: 0.5s
          update_interval: 0.5s
          min_brightness: 0%
          max_brightness: 100%
```

KVinS, 2024-09-26:
> I don't see the pipeline: section for your microphone.

I use a library that listens for a keyword in a loop:

```
micro_wake_word:
  on_wake_word_detected:
```

but it doesn't return to this cycle (

formatBCE, 2024-09-27:
> > I don't see the pipeline: section for your microphone.
> 
> I use a library that listens for a keyword in a loop:
> 
> ```
> micro_wake_word:
>   on_wake_word_detected:
> ```
> 
> but it doesn't return to this cycle (

Nothing to do with incorrect YAML. Add pipeline to your mic. It's missing.

aarongorka, 2024-11-09:
Seeing something similar, I think I narrowed it down to when you're using a `platform: adf_pipeline` media player but you use a `platform: i2s_audio` microphone. I did this because I couldn't get any results from `adf_pipeline` microphone, but `i2s_audio` works fine.

This issue seems to be fixed on `dev-next` branch, or at least it doesn't spam warnings any more.

```yaml
i2s_audio:
  - id: "my_i2s_speaker"
    i2s_lrclk_pin: GPIO21 # LRC
    i2s_bclk_pin: GPIO22
  - id: "my_i2s_microphone"
    i2s_lrclk_pin: GPIO16 # WS
    i2s_bclk_pin: GPIO4 # SCK

# # MAX98357
# speaker:
#   - platform: i2s_audio
#     id: "my_speaker"
#     dac_type: external
#     i2s_dout_pin: GPIO23
#     i2s_audio_id: my_i2s_speaker

# INMP441
microphone:
  - platform: i2s_audio
    id: my_microphone
    adc_type: external
    i2s_din_pin: GPIO0 # SD
    i2s_audio_id: my_i2s_microphone
    pdm: False
    # channel: left
    channel: right
    # bits_per_sample: 24bit
    bits_per_sample: 32bit
    sample_rate: 16000

external_components:
  - source:
      type: git
      url: https://github.com/gnumpi/esphome_audio
      ref: main
      # ref: dev-next
    components: [ adf_pipeline, i2s_audio ]

adf_pipeline:
  - platform: i2s_audio
    type: audio_out
    id: adf_i2s_out
    i2s_audio_id: my_i2s_speaker
    i2s_dout_pin: GPIO23
#   - platform: i2s_audio
#     type: audio_in
#     id: adf_i2s_in
#     i2s_audio_id: my_i2s_microphone
#     i2s_din_pin: GPIO4
#     pdm: False
#     channel: left
#     # sample_rate: 16000
#     # bits_per_sample: 32bit

# microphone:
#   - platform: adf_pipeline
#     id: my_microphone
#     # gain_log2: 3
#     # keep_pipeline_alive: True
#     pipeline:
#       - adf_i2s_in
#       - self

media_player:
  - platform: adf_pipeline
    id: my_media_player
    name: I2S Media Player
    keep_pipeline_alive: true
    internal: false
    pipeline:
      - self
      - adf_i2s_out
```