gnumpi/esphome_audio #67: Compilation Error (m5stack Atom Echo)

jarodwsams, 2024-09-07:
Let me start off by saying that I do not claim to be an expert of any sort with this stuff. I'm really good at researching and making copy/paste Frankenstein monsters.

I'm trying to incorporate the ADF Pipeline and Media Player (m5stack-atom-echo-adf.yaml) into my existing Atom Echo config, which is based on the examples from ESPHome's [Wake Word Voice Assistants repo](https://github.com/esphome/wake-word-voice-assistants/blob/main/m5stack-atom-echo/m5stack-atom-echo.yaml). My own updated configuration that I'm trying to deploy is below.

The config I've written validates ok, and starts the compile/deploy process. After almost 4 minutes, I get an error that I can't find good information on and that I do not understand. I'm hoping someone here can help me identify and resolve the issue.

I'm trying to add the media player component to my existing config so I can keep all the latest updates from ESPHome's sample config (microWakeWord and timers) and have the media player for TTS notifications.

One other thing I'll note. The config that's failing has the `esp_adf` component commented out. I ran into earlier issues that were resolved by doing this, and I assumed it would be ok since the samples here do not include that component in the config.

## Error First

> Compiling .pioenvs/my-device-name/driver/sdmmc_host.o
Compiling .pioenvs/my-device-name/driver/sdmmc_transaction.o
Compiling .pioenvs/my-device-name/driver/touch_sensor_common.o
Compiling .pioenvs/my-device-name/driver/esp32/touch_sensor.o
Compiling .pioenvs/my-device-name/driver/adc.o
Compiling .pioenvs/my-device-name/driver/esp32/adc.o
*** [.pioenvs/my-device-name/src/main.o] Error 1
========================= [FAILED] Took 272.00 seconds =========================

## My Config

Where config from the atom echo adf example replaced current config, I added line comments so I could easily compare in place.

```yaml
substitutions:
  name: my-device-name
  friendly_name: My Device Name
  micro_wake_word_model: hey_jarvis

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  min_version: 2024.7.1

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

logger:

api:
  encryption:
    key: !secret ha_encryption_key

ota:
  - platform: esphome
    id: ota_esphome
    password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  on_connect:
    - delay: 5s
  ap:
    ssid: !secret fallback_ap_ssid
    password: !secret fallback_ap_password

captive_portal:

button:
  - platform: factory_reset
    id: factory_reset_btn
    name: Factory reset

[... 383 lines cut ...]

  - source: github://jesserockz/esphome-components
    components: [file]
    refresh: 0s

  - source:
      type: git
      url: https://github.com/gnumpi/esphome_audio
      ref: main
    components: [ adf_pipeline, i2s_audio ]

#esp_adf:

file:
  - id: timer_finished_wave_file
    file: https://github.com/esphome/wake-word-voice-assistants/raw/main/sounds/timer_finished.wav

micro_wake_word:
  on_wake_word_detected:
    - voice_assistant.start:
        wake_word: !lambda return wake_word;
  vad:
  models:
    - model: ${micro_wake_word_model}
```

Rep-Hueman, 2024-09-07:
My working version: https://github.com/Rep-Hueman/ESPHome

jarodwsams, 2024-09-08:
I think I found the issue. After clearing build files and trying again, I get the following. The message is pretty clear, just not quite sure off the top of my head how to adjust.

> /config/esphome/office-voice-assist-b83ea8.yaml: In lambda function:
/config/esphome/office-voice-assist-b83ea8.yaml:215:25: error: 'class esphome::esp_adf::ADFMediaPlayer' has no member named 'play'
           - delay: 1s