gnumpi/esphome_audio #68: Will this be able to stream music to my esp32 s3 device?

manalotobry, 2024-09-17:
Hi, first of all thanks for the code - truly appreciate it. 

May I ask if there's a way to stream or play any audio to the speakers? So far, I can play TTS on its media player but no luck with streaming or playing audio.

formatBCE, 2024-09-26:
What's wrong with it? I used media_player with Music Assistant, and it works...

manalotobry, 2024-09-28:
it doesnt play any music, but it does play TTS

formatBCE, 2024-09-28:
@manalotobry just checked - my audio config is identical. And my 2 satellites are playing music.... Although I don't use them for music, they're too weak.

But Mischa said yesterday, that with new ESPHome modules, that will be released next month, his adf pipeline is redundant... Will check and see.