gnumpi/esphome_audio #74: ESPHome 2024.12.0b1 - compile error

BigBobbas, 2024-12-15:
Hi Mischa, 
I hope you are well 👍 
I'm getting the following errors when compiling, is this something you would be able to help with please.
```yaml
In file included from src/esphome/components/adf_pipeline/adf_audio_process.h:5,
                 from src/esphome/components/adf_pipeline/adf_audio_process.cpp:1:
src/esphome/components/adf_pipeline/adf_audio_element.h:6:10: fatal error: audio_element.h: No such file or directory

***********************************************************************
* Looking for audio_element.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:audio_element.h"
* Web  > https://registry.platformio.org/search?q=header:audio_element.h
*
***********************************************************************

    6 | #include <audio_element.h>
      |          ^~~~~~~~~~~~~~~~~
In file included from src/esphome/components/adf_pipeline/adf_audio_element.cpp:1:
src/esphome/components/adf_pipeline/adf_audio_element.h:6:10: fatal error: audio_element.h: No such file or directory

***********************************************************************
* Looking for audio_element.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:audio_element.h"
* Web  > https://registry.platformio.org/search?q=header:audio_element.h
*
***********************************************************************

    6 | #include <audio_element.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
compilation terminated.
*** [.pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/adf_audio_process.cpp.o] Error 1
*** [.pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/adf_audio_element.cpp.o] Error 1
In file included from src/esphome/components/adf_pipeline/adf_audio_sinks.h:5,
                 from src/esphome/components/adf_pipeline/adf_audio_sinks.cpp:1:
src/esphome/components/adf_pipeline/adf_audio_element.h:6:10: fatal error: audio_element.h: No such file or directory

***********************************************************************
* Looking for audio_element.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:audio_element.h"
* Web  > https://registry.platformio.org/search?q=header:audio_element.h
*
***********************************************************************

    6 | #include <audio_element.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/adf_audio_sinks.cpp.o] Error 1
In file included from src/esphome/components/adf_pipeline/adf_audio_sources.h:7,
                 from src/esphome/components/adf_pipeline/adf_audio_sources.cpp:1:
src/esphome/components/adf_pipeline/adf_audio_element.h:6:10: fatal error: audio_element.h: No such file or directory

***********************************************************************
* Looking for audio_element.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:audio_element.h"
* Web  > https://registry.platformio.org/search?q=header:audio_element.h
*
***********************************************************************

    6 | #include <audio_element.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/esp32-s3box-3/src/esphome/components/adf_pipeline/adf_audio_sources.cpp.o] Error 1
========================= [FAILED] Took 17.09 seconds =========================
```

gnumpi, 2024-12-16:
I guess it's due to changing the recommended idf-version to 5.1.5 in (https://github.com/esphome/esphome/pull/7706).
ADF-Pipeline does not support 5.X, you will have to set it manually back to 4.4.8 in your config yaml.
I hope this solves the issue.

BigBobbas, 2024-12-18:
Thank you for the reply , defining the following has worked.
```yaml
  framework:
    type: esp-idf
    version: 4.4.8
    platform_version: 5.4.0
```