gnumpi/esphome_audio #82: INFO ESPHome 2025.5.0 install

Homepc11qkr, 2025-05-22:
Install esphome-480-kitchen.yaml
INFO ESPHome 2025.5.0
INFO Reading configuration /config/esphome/esphome-480-kitchen.yaml...
WARNING Using `media_player.MEDIA_PLAYER_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `media_player.media_player_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: esp_adf
WARNING GPIO45 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Asia/Seoul'
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf.git@v2.5
INFO Updating submodules (components/esp-adf-libs, components/esp-sr) for https://github.com/espressif/esp-adf.git@v2.5
INFO Compiling app...
Processing esphome-480-kitchen (board: esp32-s3-devkitc-1; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
/data/cache/platformio/packages/framework-espidf@src-f61644d6990eba6c2084961ff40be733 Already Patched: yes
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.50106.0 (5.1.6) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 12.2.0+20230208 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
-- Building ESP-IDF components for target esp32s3
-- Configuring incomplete, errors occurred!
See also "/data/build/esphome-480-kitchen/.pioenvs/esphome-480-kitchen/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf@src-f61644d6990eba6c2084961ff40be733/tools/cmake/build.cmake:266 (message):
  Failed to resolve component 'esp_adc_cal'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf@src-f61644d6990eba6c2084961ff40be733/tools/cmake/build.cmake:302 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf@src-f61644d6990eba6c2084961ff40be733/tools/cmake/build.cmake:309 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf@src-f61644d6990eba6c2084961ff40be733/tools/cmake/build.cmake:603 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf@src-f61644d6990eba6c2084961ff40be733/tools/cmake/project.cmake:604 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 4.07 seconds ==========================

Homepc11qkr, 2025-05-22:
external_components:
  - source:
      type: git
      url: https://github.com/gnumpi/esphome_audio
      ref: pull/79/head
      # ref: main
    components: [adf_pipeline, i2s_audio]
    
i2s_audio:
  - id: i2s_spk
    i2s_lrclk_pin: GPIO2
    i2s_bclk_pin: GPIO1

adf_pipeline:
  - platform: i2s_audio
    type: audio_out
    id: adf_i2s_out
    i2s_audio_id: i2s_spk
    i2s_dout_pin: GPIO40
    sample_rate: 48000
    bits_per_sample: 16bit

media_player:
  - platform: adf_pipeline
    id: va_speaker
    name: Media Player
    internal: false
    keep_pipeline_alive: true
    pipeline:
      - self
      - adf_i2s_out
      - resampler

ffalsk, 2025-06-03:
Hi, I think you can try what I am commented in this: #83