gnumpi/esphome_audio #84: [ESPHome 2025.6.1] TypeError in audio schema validation: 'NoneType' for max_channels during adf_pipeline initialization

eph2, 2025-06-24:
# [ESPHome 2025.6.1] TypeError in audio schema validation: 'NoneType' for max_channels during adf_pipeline initialization (gnumpi/esphome_audio)

## ESPHome version:
2025.6.1 (running as Home Assistant Supervisor add-on)

## External component version:
gnumpi/esphome_audio - branch: `main` (latest commit as of today)

## Hardware:
ESP32-WROVER (ESP-IDF framework)

---

## Error log:

```
TypeError: unsupported operand type(s) for -: 'NoneType' and 'int'
File "/esphome/esphome/components/audio/__init__.py", line 146, in validate_audio_compatibility
max=audio_config.get(CONF_MAX_CHANNELS) - 1,
```

This happens during config validation when running `esphome compile` from ESPHome 2025.6.1.

---

## Steps already tried:

- Changed `external_components -> ref` to `main`, `dev-next`, and also tried specific SHA commits from older branches.
- Forked gnumpi/esphome_audio and tried injecting a manual `audio_device = {"max_channels": 2}` inside `adf_pipeline/__init__.py`.
- Added a manual `audio:` block in YAML with:

```yaml
audio:
  input_channels: 1
  output_channels: 1
```
Result:

```
[audio]: [output_channels] is an invalid option for [audio].
```

- Tried `max_channels` under `i2s_audio:` → Failed.
- Tried `max_channels` under each `adf_pipeline` element → Failed.
- Modified the Python call to `audio.final_validate_audio_schema()` to explicitly include `audio_device=audio_config()`. Then received:

```
TypeError: final_validate_audio_schema() missing 1 required keyword-only argument: 'audio_device'
```

- Removed and recloned the `external_components` directory multiple times.
- Used `refresh: 0s` to force ESPHome to pull fresh content.

---

## Analysis:

ESPHome 2025.6.x introduced a **mandatory `max_channels` check** inside `/esphome/components/audio/__init__.py` when validating audio components.  
The current gnumpi/esphome_audio (branch `main`) does not pass this new parameter when calling `final_validate_audio_schema()`, leading to a crash at config parse stage.

---

## Proposed Fix (for repository maintainers):

- Update `adf_pipeline/__init__.py` to pass the required `audio_device` argument when calling `final_validate_audio_schema()`.
- Example fix could be adding:

```python
audio.final_validate_audio_schema(ADF_PIPELINE_ELEMENT_SCHEMA, audio_device={"max_channels": 2})
```

- Also check for compatibility with the new ESPHome `audio:` YAML schema (introduced post-May 2025).

---

## Expected behavior:
Allow successful validation and compilation of ESPHome configs using:

- ESP-IDF framework
- adf_pipeline
- micro_wake_word
- media_player (ADFs)
- Wake word on-device
- I2S input/output

---

Thanks for maintaining this great component!  
If you need, I can help test any future fix.

canijocor, 2025-07-24:
Tengo el mismo problema, alguien sabe si hay alguna versión funcional? Acabo de recibir el m5stack coreS3 y aún no lo he podido estrenar. 😩

Ackmaniac, 2025-07-28:
Same problem here